picolibc/picolibc #2: Building for arm-none-eabi needs cc

ooxi, 2019-09-25:
First of all, thank you very much for your effort! Having struggled with newlib in the past, especially it's build system and the scattered source, this library seems like a new hope for bare metal systems :-)

I tried building picolibc for arm-none-eabi on Fedora 30 and installed arm-none-eabi-gcc. However meson tells me it needs cc, too:

```
$ ../do-arm-configure 
WARNING: Unknown CPU family , please report this at https://github.com/mesonbuild/meson/issues/new
WARNING: Unknown endian 
The Meson build system
Version: 0.50.1
Source dir: /picolibc
Build dir: /picolibc/build
Build type: cross build
Project name: picolibc
Project version: 1.0.0

meson.build:35:0: ERROR: Unknown compiler(s): ['cc', 'gcc', 'clang', 'pgcc']
The follow exceptions were encountered:
Running "cc --version" gave "[Errno 2] No such file or directory: 'cc': 'cc'"
Running "gcc --version" gave "[Errno 2] No such file or directory: 'gcc': 'gcc'"
Running "clang --version" gave "[Errno 2] No such file or directory: 'clang': 'clang'"
Running "pgcc --version" gave "[Errno 2] No such file or directory: 'pgcc': 'pgcc'"

A full log can be found at /picolibc/build/meson-logs/meson-log.txt
```

Since I'm not familiar with meson, this might be a none issue. However I'm surprised I need a host compiler when cross compiling.

Maybe we could add more detailed build instructions (with dependencies) in the documentation or provide a docker container?

keith-packard, 2019-09-26:
Thanks for reporting this issue. It looks like meson always requires a compiler for the build machine in addition to the machine for which the code will be compiled. I've filed an issue against the meson build system, https://github.com/mesonbuild/meson/issues/5962, but for now, picolibc will require both native and cross compilers to run.

keith-packard, 2019-09-29:
I'm going to just close this issue; I read through the meson source code and I'm reasonably sure there's no way to configure picolibc to not require a compiler for the build machine too. If they ever change things, then picolibc will magically start compiling without that.

closed 2019-09-29 by keith-packard

ooxi, 2019-10-11:
Thanks for the feedback anyway! :-)