picolibc/picolibc #3: unknown option "optimization" was reported while do-arm-configure

jserv, 2019-09-28:
I managed to build picolibc with [GNU Arm Embedded Toolchain](https://launchpad.net/gcc-arm-embedded) and suffered from Meson build problems:
```
$ ../do-arm-configure 
The Meson build system
Version: 0.45.1
...
Cross C compiler: arm-none-eabi-gcc (gcc 7.3.1)
...
meson.build:79:0: ERROR: Tried to access unknown option "optimization".
```

Line 79 of file `meson.build`:
```
target_optspace = get_option('optimization') == 's'
```

It looks inconsistent to other entries in the same file.

keith-packard, 2019-09-28:
You're using a version of meson which doesn't compute this built-in option from the build type. The 'optimization' option was added in meson version 0.48.0. I'll switch this to using the 'buildtype' option instead of the optimization option and test against an older meson to see if that works.

jserv, 2019-09-28:
I am using Ubuntu Linux 18.04-LTS and `messon` package is of version `0.45.1`. Thus, I removed `meson` package and [install it via pip3](https://mesonbuild.com/Getting-meson.html).
```
pip3 install meson
```

Now, it works as expected.
```
$ ../do-arm-configure 
The Meson build system
Version: 0.51.2
...
Message: Computed host_cpu as arm
Compiler for C supports function attribute alias: YES 
Configuring picolibc.specs using configuration
Checking if "long double check" compiles: YES 
Compiler for C supports arguments -fno-tree-loop-distribute-patterns: YES 
Configuring newlib.h using configuration
Configuring _newlib_version.h using configuration

$ ninja
10743/17176] Compiling C object 'newlib/libm/common/cf42e1f@@common@sta/nexttowardf.c.o'.
../newlib/libm/common/nexttowardf.c: In function 'nexttowardf':
...
```

@keith-packard, would you add notice about Meson versioning for build instructions?

keith-packard, 2019-09-29:
Thanks, I've added a meson version check to the build system; it will now complain if you attempt to build with too old a meson version. I looked at making it build with 0.45, but I'm using several features newer than that, the newest being dictionaries which weren't added until 0.48. I tested with that version and it works correctly.

closed 2019-09-29 by keith-packard