picolibc/picolibc #5: library 'm' not found when 'tests' is set to true

jserv, 2019-09-28:
I tried to append `-Dtests=true` to meson when configuring picolibc. However, it failed:
```
newlib/libm/test/meson.build:125:0: ERROR: C library 'm' not found
```

After checking `meson-logs/meson-log.txt`, I figured out the corresponding parts:
```
Adding test "math"
Running compile:
Working directory:  /tmp/tmpaukv19l_
Command line:  arm-none-eabi-gcc /tmp/tmpaukv19l_/testfile.c -o /tmp/tmpaukv19l_/output.exe \
-pipe -D_FILE_OFFSET_BITS=64 -O0 \
-Wl,--start-group -lm -Wl,--end-group -Wl,--allow-shlib-undefined

Code:
 int main() { return 0; }
Compiler stdout:

Compiler stderr:
 /usr/bin/../lib/gcc/arm-none-eabi/7.3.1/../../../../arm-none-eabi/lib/libc.a(lib_a-exit.o): In function `exit':
exit.c:(.text.exit+0x2c): undefined reference to `_exit'
collect2: error: ld returned 1 exit status
```

keith-packard, 2019-09-29:
I'm afraid all of the tests only currently run when the library is compiled to run on Linux and not an embedded system. That would be great to fix, but it would require a target device for testing, and the current picolibc bits don't have any target-specific code.

Try the 'do-native-configure' or 'do-x86-configure' scripts; those will at least build the tests for those targets and provide some level of testing of the library code there.

What I hope to do is to build tests for qemu running a RISC-V emulation with freedom-metal hardware support. That would offer the ability to test a wide range of target architectures. Doing something similar for ARM would also be a fabulous plan.

keith-packard, 2020-01-02:
This should all be working now; picolibc can now build (and execute) tests using QEMU on both RISC-V (qemu bits not upstream yet) and ARM.

closed 2020-01-02 by keith-packard