picolibc/picolibc #6: Get rid of GPL'd printf tests

jserv, 2019-09-28:
The `printf` test suite might be the only source licensed under GNU GPL. Why not adopting other test cases without GPL? The [test suite](https://github.com/mpaland/printf/tree/master/test) from [mpaland/printf](https://github.com/mpaland/printf) looks pretty well and it is MIT licensed.

keith-packard, 2019-09-29:
I'm fine with GPL code included as part of the test suite; that isn't delivered in applications and hence has no effect on overall target licensing. The printf test suite I included was gathered from a huge range of existing test suites and covers a nice array of corner cases, so I'd like to keep it around.

joshtriplett, 2020-01-07:
@keith-packard That seems inconsistent with the licensing information in the README, which says that "none of the GPL-related bits are left in the repository". (I'm fine with GPLed code too, but the README should match the current status.)

keith-packard, 2020-01-07:
Good point. I'll update the README to reflect the status of the printf tests

keith-packard, 2020-05-07:
COPYING.picolibc now reflects all of the licenses in use, README.md notes that there are some tests under the GPL which do not end up in the library.

closed 2020-05-07 by keith-packard