picolibc/picolibc #8: Add option to intallation prefix in gcc sysroot

martinribelotta, 2019-10-05:
In some cross-builds, is desirable to set the installation path (prefix) into the gcc-sysroot (by example `${GCC PREFIX}/arm-none-eabi/`

This may be implemented using the command:

```sh
${CORSS_PREFIX}gcc --print-sysroot
```

I could make the changes on my fork, but I would like to discuss "the how"

keith-packard, 2019-10-06:
Yeah, if you'll look at how the build currently works, it only sticks the 'picolibc.specs' file in the GCC installation directory. With that, you can use '-specs=picolibc.specs' and the library and include paths will get set when you compile. The extra bonus feature is that the library path will point at the right architecture given the various -m options passed in to gcc.

If you want to *also* install picolibc in the gcc directory, you could do that, but I'm not sure we really want that? glibc doesn't do that, for instance.

Do you not want to use -specs=picolibc.specs? Or is there something I'm just missing here?

martinribelotta, 2019-10-06:
The idea behind install libc in gcc directory is provide a complete distributable archive (like a current newlib distribution or many cross toolchain based on glibc or musl)

In many self-build cross-toolchains I prefer to integrate the libc (musl, glibc, etc) into gcc directory for distribution simblicity.

I have tried manually installing picolibc on gcc-sysroot and modifying the specfile and it works fine and not break the existing newlib installation (at least in arm-gcc-embedded)

Install libc into gcc bundle is not the desired behabioural if you use system gcc like from a debian package of gcc-arm-none-eabi but is a good idea if you try to distribute the compiler as a zip/tgz file. 
Therefore, this option should not be active by default.

keith-packard, 2019-10-08:
Sounds like a useful option then. Please feel free to try and add this if you can.

closed 2019-11-18 by martinribelotta