picolibc/picolibc #11: Deprecated machines for picolibc?

jserv, 2019-10-07:
At present, there are 57 machines listed in directory `newlib/libc/machine/`. But I have no idea how many targets of them are really supported by picolibc along with `picocrt`. Is it worth to leave unsupported targets?

mithro, 2019-10-07:
I would actually love to use picolibc on our FPGA soft cores and thus love to see the following supported (in order of my priority);
 * rv32
 * rv64
 * ppc64le
 * arm
 * ppc32
 * arm64
 * or1k
 * lm32
 * sh2 + sh4

keith-packard, 2019-10-08:
I'd like to leave the library code for these architectures around -- it makes merging newlib changes easier so that we can be prepared for supporting those when possible. Adding support for a new architecture for picolibc is mostly a matter of creating the meson build files. Testing those requires a suitable toolchain, so if you've got compilers for your target architectures, you should be able to add the meson bits pretty quickly.

I've added the list of currently supported architectures to the README.md, along with a short description of the tasks necessary to add more architecture support. I think that's sufficient for now.

closed 2019-10-08 by keith-packard