picolibc/picolibc #19: No specs files?

antmak, 2019-11-18:
You mean specs files https://github.com/keith-packard/picolibc/blob/master/hello-world/README.md but no ones in the repo.

keith-packard, 2019-11-18:
Thanks for reporting an issue; I hope we just need to clarify the language in the README.md.

hello-world/README.md is referring to the picolibc.specs file which is generated during the build of picolibc from [picolibc.specs.in](https://github.com/keith-packard/picolibc/blob/master/picolibc.specs.in). 

The hello world sample is supposed to show how to use picolibc *after* it has been installed; it won't work from an uninstalled tree. Does this help at all? Do you have suggestions on how to clarify the hello world README file?

antmak, 2019-11-18:
Ah, I had missed the picolibc.specs.in file when was taking a look on the lib. Ok, thanks for showing. Maybe you can add a small notice that _"Picolibc provides a GCC '.specs' file __(generated form picolibc.specs.in)__ which sets the search path for header files and picolibc libraries."_  just to have a quick way to get the content, for inattentive people like me :)

keith-packard, 2019-11-18:
Good idea. I've gone ahead and done that.

closed 2019-11-18 by keith-packard