picolibc/picolibc #20: picolibc build.md documentation needs to be updated with correct info (doc -> build.md)

jedxops, 2019-11-21:
The file is located at the following directory from this github repo
    doc/build.md
Please see the following screenshot:

[image]

The cross-riscv64-unknown-elf.txt file should have values encapsulated by the ''
And it should be using 64 bit values, not 32. So all the 32's need to be replaced with 64.
Also, the filename itself is wrong, should be `cross-riscv64-unknown-elf.txt`

Please fix to avoid a mountain of confusion from readers.

keith-packard, 2019-11-21:
Thanks! All fixed on master.

closed 2019-11-21 by keith-packard

jedxops, 2019-11-21:
> Thanks! All fixed on master.
Doesnt look fixed to me, I am on master branch. Nothing is different!

jedxops, 2019-12-03:
much better thanks!