picolibc/picolibc #23: Picocrt ARM Floating Point Initialization

sdonnan, 2020-01-14:
I am targeting the stm32l4 with the following compiler settings:

```
-mthumb23
-march=armv7e-m
-mfloat-abi=hard
-mfpu=fpv4-sp-d16
```

The picocrt code in `crt0.c` has some assembly to set the "flush-to-zero" bit in the `FPSCR`. However it attempts to do this before enabling the floating point unit. This is required per the Cortex M4 reference manual section 7.3.2.

My work around was to add the following before the flush-to-zero bit is set
```
#define CPACR 0xE000ED88
#define CPACR_CP10_FULL (0x3 << 10*2)
#define CPACR_CP11_FULL (0x3 << 11*2)
 
// Enable full access to coprocessor 10 and 11 for hardware floating point
*(int*)CPACR |= CPACR_CP10_FULL | CPACR_CP11_FULL; 
```

However it appears this is only necessary on Cortex-M4 parts. Other M series parts I checked (M7, M33) do not appear to need the initialization.

Could we remove the flush-to-zero bit setting code and leave it to the user to configure the FPU in the `_init` function (or init array) so they can control the ordering properly for their platform?

sdonnan, 2020-01-14:
I see now that one of the goals of picocrt is to setup proper rounding behavior in the FPU. I'm not sure what the right approach is to handle this case.

To further muddy the waters, at least on my test board, there must be some delay between enabling the FPU and accessing its registers. So it is a pretty ugly initialization.

At this point it probably makes sense to just implement the crt myself and not link against picocrt.

keith-packard, 2020-01-15:
Ideally, picocrt would be able to do all of the initialization necessary so that it could actually work without needing the application to help out. If you've got some code that can do this, I'd love to get it integrated. I've got several M4 boards to test with. If this turns out to be impossible (I hope not), then, yes, we should pull the code which will crash the board out of the library.

Thanks much for getting this far and figuring out what the code doing wrong.

sdonnan, 2020-01-15:
Sounds good. I'll take another pass at integrating it and see if I can come up with something reasonable. My current approach uses a busy wait, which works for my application, but likely isn't portable.

keith-packard, 2020-07-10:
picocrt now has ARM FPU initialization code, but I haven't managed to test it on real hardware. I *think* it should work though -- it enables the FPU and then spins until it appears to be initialized. I'm going to close this; if you get a chance to give that a try, please let me know if it works fo ryou.

sdonnan, 2020-07-15:
My initial check of with an STM32L4 and commit `b33afe886` did not work. I'll get it under the debugger soon and track down what is going on.

keith-packard, 2020-07-15:
thanks, I'll re-open the bug then. alas.

zip639, 2021-02-25:
Don't know if this will help but ARM gives a sequence of assembly instructions to enable the FPU on Cortex-M, you might want to try it (it looks kinda close to what you're already doing through): 
```
CPACR   EQU  0xE000ED88
LDR     R0,  =CPACR          ; Read CPACR
LDR     r1, [R0]             ; Set bits 20-23 to enable CP10 and CP11 coprocessors
ORR     R1, R1, #(0xF << 20)
STR     R1, [R0]             ; Write back the modified value to the CPACR
DSB
ISB                          ; Reset pipeline now the FPU is enabled.
```

keith-packard, 2021-02-26:
thanks for the updated reference -- the ARM that I looked at didn't include the DSB/ISB instructions. Adding those makes this work reliably on at least one of the M4 boards I've got here. I'm closing this issue as the library now works on actual hardware.

closed 2021-02-26 by keith-packard