picolibc/picolibc #30: Portability issue with non-basic asm statements in naked functions

PkmX, 2020-02-24:
In `picocrt/machine/riscv/crt0.c`, `_start` is defined as a naked function but it also contains C declarations and statements, which according to gcc's [documentation](https://gcc.gnu.org/onlinedocs/gcc/RISC-V-Function-Attributes.html) is not reliably supported:

> While using extended asm or a mixture of basic asm and C code may appear to work, they cannot be depended upon to work reliably and are not supported.

`clang` just flat out rejects these programs:

```
crt0.c:57:2:error: non-ASM statement in naked function is not supported
        __start();
```

It seems to me that there is no reason why the offending code has to be written in C, and they can be written in pure assembly in basic asm blocks trivially.

keith-packard, 2020-02-24:
__start is defined as a static inline, so I think it should probably be called from a regular C function so that it can all get nicely inlined in a reasonable environment. Instead of changing that around, how about we make the 'naked' function, _start, as short as possible. All we need to be able to call C functions is to set up the stack. This works fine with GCC; let me know if you like this plan or not.

[0001-picocrt-Make-naked-RISC-V-_start-contain-only-asm-st.txt](https://github.com/keith-packard/picolibc/files/4243394/0001-picocrt-Make-naked-RISC-V-_start-contain-only-asm-st.txt)

PkmX, 2020-02-24:
That patch should work. Although I'd prefer to have `_start` handle all inline assembly sequences needed to initialize the environment expected by a C function, with a jump to `__start` directly at the end. As for the static inline thing, I don't think it is possible to achieve good inlining between naked/normal functions using standard gcc features, so there has to be a `j` somewhere unless we just resort to writing the whole sequence in assembly.

keith-packard, 2020-02-24:
I considered that, but then I'd have to change the common code shared between RISC-V and other architectures so that __start was not inline. Plus, we'd have more non-C code, which I like to avoid, given that C code is easier to read than asm (although, the random mixture of C and asm isn't all that readable...).

In any case, as long as it works, I think I'll leave it alone. I've tested this using QEMU in 30 different configurations, which is encouraging at least :-).

Thanks for your review!

closed 2020-02-24 by keith-packard