picolibc/picolibc #34: Using picolibc from C++ result in inclusion of conflicting newlib includes

thmxv, 2020-04-11:
Hi,

I have a custom built riscv64 toolchains with 3 libc (newlib, newlib-nano and picolibc). I can choose which libc to compile my project with just by setting the --spec=<which_libc>.specs command line option. However when compiling C++ code with picolibc, the compilation fails because of conflicting redeclarations caused by the inclusion of include files from newlib, when only include files from picolibc should be found:
```
[2/13] Building CXX object kernel/CMakeFiles/kernel.dir/src/stubs/newlib_stubs.cxx.obj
FAILED: kernel/CMakeFiles/kernel.dir/src/stubs/newlib_stubs.cxx.obj 
/home/xavier/Projects/sp3/current_toolchains/riscv64-unknown-elf/bin/riscv64-unknown-elf-g++  -DFMT_HEADER_ONLY=1 -D_LITE_ENTRY -I/home/xavier/Projects/sp3/sp3/libsp3/include -I/home/xavier/Projects/sp3/sp3/third_party/fmt-6.2.0/include -mcmodel=medany --specs=picolibc.specs -ffreestanding -fno-common -ffunction-sections -fdata-sections -fno-rtti -fno-exceptions -fno-non-call-exceptions -fno-unwind-tables -fno-use-cxa-atexit  -O3 -DNDEBUG   -Wall -Wextra -Wshadow -Wnon-virtual-dtor -Wold-style-cast -Wcast-align -Wunused -Woverloaded-virtual -Wpedantic -Wconversion -Wsign-conversion -Wnull-dereference -Wdouble-promotion -Wformat=2 -Werror -Wmisleading-indentation -Wduplicated-cond -Wduplicated-branches -Wlogical-op -std=gnu++2a -MD -MT kernel/CMakeFiles/kernel.dir/src/stubs/newlib_stubs.cxx.obj -MF kernel/CMakeFiles/kernel.dir/src/stubs/newlib_stubs.cxx.obj.d -o kernel/CMakeFiles/kernel.dir/src/stubs/newlib_stubs.cxx.obj -c /home/xavier/Projects/sp3/sp3/kernel/src/stubs/newlib_stubs.cxx
In file included from /home/xavier/Projects/sp3/current_toolchains/riscv64-unknown-elf/riscv64-unknown-elf/include/stdlib.h:18,
                 from /home/xavier/Projects/sp3/current_toolchains/riscv64-unknown-elf/riscv64-unknown-elf/include/c++/9.3.0/cstdlib:75,
                 from /home/xavier/Projects/sp3/current_toolchains/riscv64-unknown-elf/riscv64-unknown-elf/include/c++/9.3.0/bits/stl_algo.h:59,
                 from /home/xavier/Projects/sp3/current_toolchains/riscv64-unknown-elf/riscv64-unknown-elf/include/c++/9.3.0/string:52,
                 from /home/xavier/Projects/sp3/current_toolchains/riscv64-unknown-elf/riscv64-unknown-elf/include/c++/9.3.0/stdexcept:39,
                 from /home/xavier/Projects/sp3/current_toolchains/riscv64-unknown-elf/riscv64-unknown-elf/include/c++/9.3.0/array:39,
                 from /home/xavier/Projects/sp3/current_toolchains/riscv64-unknown-elf/riscv64-unknown-elf/include/c++/9.3.0/tuple:39,
                 from /home/xavier/Projects/sp3/current_toolchains/riscv64-unknown-elf/riscv64-unknown-elf/include/c++/9.3.0/bits/unique_ptr.h:37,
                 from /home/xavier/Projects/sp3/current_toolchains/riscv64-unknown-elf/riscv64-unknown-elf/include/c++/9.3.0/memory:80,
                 from /home/xavier/Projects/sp3/sp3/kernel/src/stubs/../console.h:4,
                 from /home/xavier/Projects/sp3/sp3/kernel/src/stubs/newlib_stubs.cxx:1:
/home/xavier/Projects/sp3/current_toolchains/riscv64-unknown-elf/riscv64-unknown-elf/include/sys/reent.h:292:26: error: conflicting declaration 'typedef struct __sFILE __FILE'
  292 | typedef struct __sFILE   __FILE;
      |                          ^~~~~~
In file included from /home/xavier/Projects/sp3/current_toolchains/riscv64-unknown-elf/riscv64-unknown-elf/sysroot/picolibc/riscv64-unknown-elf/include/wchar.h:31,
                 from /home/xavier/Projects/sp3/current_toolchains/riscv64-unknown-elf/riscv64-unknown-elf/include/c++/9.3.0/cwchar:44,
                 from /home/xavier/Projects/sp3/current_toolchains/riscv64-unknown-elf/riscv64-unknown-elf/include/c++/9.3.0/bits/postypes.h:40,
                 from /home/xavier/Projects/sp3/current_toolchains/riscv64-unknown-elf/riscv64-unknown-elf/include/c++/9.3.0/iosfwd:40,
                 from /home/xavier/Projects/sp3/current_toolchains/riscv64-unknown-elf/riscv64-unknown-elf/include/c++/9.3.0/memory:72,
                 from /home/xavier/Projects/sp3/sp3/kernel/src/stubs/../console.h:4,
                 from /home/xavier/Projects/sp3/sp3/kernel/src/stubs/newlib_stubs.cxx:1:
/home/xavier/Projects/sp3/current_toolchains/riscv64-unknown-elf/riscv64-unknown-elf/sysroot/picolibc/riscv64-unknown-elf/include/stdio.h:266:23: note: previous declaration as 'typedef struct __file __FILE'
  266 | typedef struct __file __FILE;
```
I am pretty sure this is caused by the use of `#include_next <stdlib.h>` in `include/c++/9.3.0/cstdlib`

keith-packard, 2020-04-11:
I think the underlying issue is that you've got the newlib headers installed in the default include search path, so that when cstdlib does this:
```
#define _GLIBCXX_INCLUDE_NEXT_C_HEADERS
#include_next <stdlib.h>
#undef _GLIBCXX_INCLUDE_NEXT_C_HEADERS
```
instead of getting the picolibc stdlib.h, you end up getting the newlib stdlib.h

I *think* this can be fixed by getting the newlib headers out of the GCC search path, but GCC is
pretty sure that you never want to do this as it makes it really hard. The only way to remove the default libc header directory is to use '-nostdinc', but that also removes the GCC internal header directory, so you have to re-add that with -isystem. On my debian system which is using the packaged arm-none-eabi toolchain, I did this:
```
arm-none-eabi-gcc --specs=picolibc.specs -mcpu=cortex-m3 -Thello-world-arm.ld -nostdinc -isystem /usr/lib/gcc/arm-none-eabi/8.3.1/include -o hello-world-arm.elf hello-world.c
```
Replace /usr/lib/gcc/arm-none-eabi/8.3.1/include with the path to the GCC internal header files on your system (files like float.h, stdint.h and friends).

There is a larger problem which you will probably run into -- the standard C++ library is part of gcc, and is built against one of the three C libraries you've got, almost certainly newlib. When you use functions from that library which reference stdio from libc, you'll be using the newlib ABI, not the newlib-nano or picolibc ABI. We need to come up with a way to build this library against another libc and make that available as part of the newlib-nano or picolibc environment.

If you just want to switch to picolibc (or newlib-nano), and abandon newlib entirely, it should be possible to fix gcc to build against picolibc/newlib-nano instead of newlib, and then install that libc in the 'standard' path so that you wouldn't need --specs= at all.

thmxv, 2020-04-12:
Thanks for the pointer to -nostdinc, I did not think of it, but it should work indeed. I will test and report back on my (lack of) success. Is there a way to do this reliably in the spec file?

Regarding libstdc++ being built against newlib is a problem I am aware of. I don't link libstdc++ on my project (and don't plan to), I just use some headers. When I try to build picolibc with a "stage 1 gcc" (without libc) the build fails. For now it only compiles with the "stage 2 gcc" that was build with newlib. I need to do more work to find what I am doing wrong, but worst case I can built gcc a third time.

keith-packard, 2020-04-12:
-nostdinc seems to work fine in the *cpp clause of a .specs file, so you can add
```
*cpp
-nostdinc -isystem <path-to-gcc>/include -isystem <path-to-picolibc>/inc %(picolibc_cpp)
```
Getting this into the picolibc.specs file generated by the build system might be a bit tricky as it also needs to handle the sysroot configuration option.

I've spent a bit of time today working on building libstdc++ for picolibc; it looks feasible, so perhaps I'll see about forking gcc and hacking it up to create a stand-alone libstdc++ project.

thmxv, 2020-04-12:
I do use a version of picolibc  configured with the `-Dsysroot-install=true` and with the install prefix set to sysroot. I had to add few more directories with  `-isystem`, as showed in the cpp line of my `picolibc.specs`:
```
*cpp:
-nostdinc -isystem %R/../../lib/gcc/riscv64-unknown-elf/9.3.0/include -isystem %R/../../lib/gcc/riscv64-unknown-elf/9.3.0/include-fixed -isystem %R/../include/c++/9.3.0 -isystem %R/../include/c++/9.3.0/riscv64-unknown-elf -isystem %R/picolibc/riscv64-unknown-elf/include %(picolibc_cpp)
```
Now I have an `error: '::strtold' has not been declared ` but this is unrelated to this issue and I probably just need to recompile using `-Dnewlib-io-long-double=true`

thmxv, 2020-04-12:
OK, so this is not really related to the original issue title, and you are probably light years ahead of me and figured this out already but I keep reporting here anyway. If you wish for me to stop reporting on this or open another issue, feel free to tell me.

I think the errors I have  are not due to the configure options of picolibc but rather to the fact that some of libstdc++ headers are generated based on the content of some libc headers. So using a libstdc++ compiled against newlib will not work with picolibc even if I only include some headers an do not link libstdc++. So the idea of switching from newlib to picolibc by the use of a spec file will not work in C++ like it does in C. Rather I will need to build 2 different toolchains and I will need to build libstdc++ for picolibc.

keith-packard, 2020-04-12:
> I think the errors I have are not due to the configure options of picolibc but rather to the fact that some of libstdc++ headers are generated based on the content of some libc headers. So using a [...]

I'm not sure  there's any dependency on the header files, but definitely inside the library; as far as I've found out so far, libstdc++ only depends on libc's stdio and ctype bits. "Fortunately", picolibc shares newlib's ctype implementation, so we should be able to build libstdc++ against picolibc using the libstdc++ bits which assume a newlib based ctype implementation. I've compiled libstdc++ against picolibc, and will be working on testing that, with an eye towards releasing a stand-alone libstdc++ that can compile against newlib, newlib-nano and picolibc, without needing an entire toolchain built. libstdc++ should never have been included in gcc...

> So the idea of switching from newlib to picolibc by the use of a spec file will not work in C++ like it does in C. Rather I will need to build 2 different toolchains and I will need to build libstdc++ for picolibc.

I hope we can get by with sharing the toolchain and distributing separate libstdc++ versions for each libc. I hope to have this working sometime this week. Thanks much for your help exploring how this might work.

thmxv, 2020-04-12:
You have more success than me. I did not managed to get libstdc++ to compile successfully against picolibc (yet).

All you say make sense. I just don't think I have the skills to make it work by myself. You effort to make  a setup like this more accessible is greatly appreciated. Thank you.

I will continue to do some tests on my side and of course I am available to test your work.

keith-packard, 2020-04-13:
> You have more success than me. I did not managed to get libstdc++ to compile successfully against picolibc (yet).

It certainly wasn't easy to manage... I haven't quite sorted out how I'm going to create a source package. I don't want to include all of gcc, but I need to keep the ability to merge in upstream changes. Suggestions welcome, of course.

> I will continue to do some tests on my side and of course I am available to test your work.

I'll be creating Debian packages that will work alongside the Debian packages for riscv64-unknown-elf gcc and picolibc. Those will (eventually) make it into Debian unstable, but if you're using Debian or a derivative, they're available in my personal debian repository:

[keithp.com/archive](https://keithp.com/archive/)

thmxv, 2020-04-13:
I am using arch linux but I can build  my own for arch after looking how the debian package is built.

I managed to compile libstdc++ using --specs=picolibc.specs and installed into sysroot/picolibc but did not managed to get a modified specs file that  work yet.

keith-packard, 2020-04-14:
I discovered today that GCC's libstdc++ depends on libc supporting the POSIX functions read, write, lseek64, etc. I've hacked it up to run without those functions. I then built it for RISC-V. Then I discovered a bunch of missing stdio functions in picolibc's tinystdio. Finally, I had to add some more stuff to picolibc.ld. The result is that I've got a sample 'hello world' C++ program running under qemu. It's a bit large; the C version is 1220 bytes, the C++ version is over 128kB. I'll clean up what I've done and see about getting some code uploaded so you can see too.

thmxv, 2020-04-14:
That is quite impressive. Thanks for the good news

The 100x size difference is a bit much, but is it using iostreams, exceptions, rtti, ...? Or is it exactly the same code but using <cstdio> and std::printf() instead of <stdio.h> and printf()? All things equivalent, and using only the "zero cost" features of C++ it should be possible to end with almost the same binary executable. 

I will definitely try it out when possible.

thmxv, 2020-04-16:
I did some tests using on a small project I am currently working on. The code use newlib_nano for the moment. It also uses a lot of c++ headers (array, algorithm, atomic, string_view, memory,  ...) in addition to the C++ified C headers (cassert, cstddef, cstdint, ...). Last but not least the code also use the famous C++ library libfmt.

The project is compiled without RTTI and exceptions. It compiles and execute fine when not linking libsupc++ and libstdc++ (it requires additional code for all the new/delete operators and a few  __throw_xxx stubs). 

When compiling for size (-Os, LTO, -function-sections, -fdata-sections, -Wl,--gc-sections) the code is 23K.

Now just adding libstdc++ as a link dependency and the code size grows by more than 100K :(

The only way I found to "fix" this is to recompile libstdc++ without exceptions (by using -fno-exceptions in addition to -mcmodel=medany  in the CXXFLAGS_FOR_TARGET)

Linking with such a libstdc++ my project's binary end up being 24K.

keith-packard, 2020-04-17:
That's good news, although of course *my* customers want exceptions enabled. I wonder if we simply need to provide two versions of libstdc++ when shipping binary packages. We could stick something in the .specs file to control which was selected. Ick.

thmxv, 2020-04-18:
That look a bit like a rabbit hole. People might want also `-fno-rtti`, which would make it 3 binary versions. Maybe other options like stack unwinding support make sense to enable/disable. 

I am afraid the standard procedure is to only provide binary that supports the standard C++, while having a way to do custom builds that do not. People, that disable some of the non-zero cost options, know they are using a "incompatible C++ dialect" and cannot link against any library using those features. I just naively thought that because my code do not use any of it the compiler/linker would optimize this way for me. 

Also I think libstdc++ have some special mechanism in place to catch any exceptions in flight at the "boundaries" between code code using exceptions or not; and that for other libraries not using such mechanism it would be an ABI incompatibility probably resulting in a crash. But don quote me on this, I am in the middle of learning all this the hard way.

keith-packard, 2020-04-20:
Ok, sounds like a reasonable first step would be to construct an external libstdc++ project and let people build that themselves, instead of attempting to build all possible variations.

keith-packard, 2021-01-28:
I've spend a bunch of time working on this little project and have discovered that the only way to make this work is to modify the headers installed by libstdc++ so that they do not use '#include_next' as that ends up using the wrong C library headers.

I've got a sample configuration for crosstool-ng which builds newlib, newlib-nano and picolibc all in the same toolchain directory and lets you use c++ from all of them.

The patches necessary to make libstdc++ build against picolibc are also upstream in gcc and should be included in gcc version 11.

With that, I'm going to close this issue as there's nothing else to be done here.

closed 2021-01-28 by keith-packard