picolibc/picolibc #36: Meson option newlib-supplied-syscalls is broken

denizzzka, 2020-05-02:
Error message:
```
subprojects/picolibc/newlib/libc/meson.build:155:2: ERROR: Non-existent build file 'subprojects/picolibc/newlib/libc/syscalls/meson.build'
```

Directory `picolibc/newlib/libc/syscalls` directory does not exist

denizzzka, 2020-05-02:
Commit 22183b26c08bd8a4c5b5b4c6f438a10700f0f1cc related to this.

I thought picolibc provides some stubs if `newlib-supplied-syscalls` used for case if system calls are not available. Is this false assumption?

keith-packard, 2020-05-02:
yeah, that's a no-longer-valid option from newlib that didn't get cleaned up. In newlib, that option hooks various POSIX functions through 'struct reent'-using versions which then call into libgloss. 

For example:
newlib/libc/syscalls/sysread.c:
```
read (int fd,
     void *buf,
     size_t cnt)
{
  return _read_r (_REENT, fd, buf, cnt);
}
```
newlib/libc/reent/readr.c:
```
_ssize_t
_read_r (struct _reent *ptr,
     int fd,
     void *buf,
     size_t cnt)
{
  _ssize_t ret;

  errno = 0;
  if ((ret = (_ssize_t)_read (fd, buf, cnt)) == -1 && errno != 0)
    ptr->_errno = errno;
  return ret;
}
```
finally, libgloss/arm/sycalls.c:
```

int __attribute__((weak))
_read (int fd, void * ptr, size_t len)
{
  int res;
  struct fdent *pfd;

  pfd = findslot (fd);
  if (pfd == NULL)
    {
      errno = EBADF;
      return -1;
    }

  res = _swiread (pfd->handle, ptr, len);

  if (res == -1)
    return res;

  pfd->pos += len - res;

  /* res == len is not an error, 
     at least if we want feof() to work.  */
  return len - res;
}
```
That whole call chain exists solely to map libgloss 'errno' into newlib's reent->_errno value.
That's a bunch of redirection we don't need in picolibc because an underlying POSIX API can just use a thread-local errno directly.

Take a look at picolibc/semihost. That offers a handful of POSIX APIs built on top of ARM/RISC-V semihosting. It implements those without the intervening _read_r and _read functions, writing error results directly into 'errno', which is declared as extern __thread int _errno.

 Hrm. I should rename that as just 'errno'; there's no reason to prefix it with an _ and use a macro for the name.

Picolibc also has 'dummyhost', which (currently) only defines __iob so that you can use stdin/stdout/stderr without causing a link failure. If you want additional no-op syscalls for some reason, we could put them in there?

denizzzka, 2020-05-03:
> Hrm. I should rename that as just 'errno'; there's no reason to prefix it with an _ and use a macro for the name.

The question may have arisen by this reason. I linking latest `picolib` version and it is available `read` function with same arguments exported from semihost, but not `_read` isn't available. This causes error:
```
newlib/libc/reent/readr.c:49: undefined reference to `_read'
```
(same error with _sbrk, _write, _close and _lseek)

This is by design and I should for example just to write some wrap functions `_read` => `read`  like it described above?

keith-packard, 2020-05-03:
Hrm. Even if you're building with the old stdio code (-Dnewlib-tinystdio=false), reent/readr.c shouldn't be getting compiled into the library; the only stuff from reent used at this point is the code for managing the 'struct reent' needed by the old stdio code. That happened before picolibc version 1.0 was released. I should remove the unused code from the library sources.

The original stdio code needs a bunch of work to eliminate use of 'struct reent' and switch to
TLS storage; I haven't done that because my target environments all use tinystdio. Do you need that older code? Should I spend time getting it moved over to TLS? It would also take a bunch of work to get it passing the test suite for printf...

You have encouraged me to at least go and see what the state of the old stdio bits was. I found a few issues with getting it running using either glibc or semihost for the POSIX functions, and I've fixed those on the master branch.

Let's figure out what code you've got and make sure you're at version 1.4.1, then we can sort out the build issues.

denizzzka, 2020-05-04:
> Hrm. Even if you're building with the old stdio code (-Dnewlib-tinystdio=false), reent/readr.c shouldn't be getting compiled into the library

I build by Meson, and only one option is set: multilib=false ('Enable multilib compile').
newlib-tinystdio=true by default, this value used

> Do you need that older code? Should I spend time getting it moved over to TLS?

So, this is not needed, thanks!

> Let's figure out what code you've got and make sure you're at version 1.4.1, then we can sort out the build issues.

I am on picolib 57d4fbe7dc336a8073727e9291e612e2879a8c1c (tagged 1.4.1)

denizzzka, 2020-05-04:
current ~master (1781d82bd69086de53578afd17099be68e0a7ad0) compilation:
```
../subprojects/picolibc/newlib/libc/tinystdio/vfprintf.c:334:40: error: function definition is not allowed here
    void my_putc(char c, FILE *stream) {
                                       ^
```

keith-packard, 2020-05-04:
> current ~master ([1781d82](https://github.com/keith-packard/picolibc/commit/1781d82bd69086de53578afd17099be68e0a7ad0)) compilation:
> 
> ```
> ../subprojects/picolibc/newlib/libc/tinystdio/vfprintf.c:334:40: error: function definition is not allowed here
>     void my_putc(char c, FILE *stream) {
>                                        ^
> ```

I was just looking at this code today; trying to figure out why it was larger than the original tinystdio vfprintf function. Some of it was just bug fixing, but quite a bit turned out to be from using a nested function instead of a simple macro. I've switched to using a macro, which should make non-GCC compilers happier too.

keith-packard, 2020-05-04:
> I build by Meson, and only one option is set: multilib=false ('Enable multilib compile').
> newlib-tinystdio=true by default, this value used

I'm a bit confused -- are you attempting to use picolibc in a Linux hosted environment, not building it using an embedded cross compiler?

> So, this is not needed, thanks!

Definitely not. I'll go remove those source files as they will never be needed again.

Thanks for giving this all a try; it's good to get feedback!

denizzzka, 2020-05-04:
> I've switched to using a macro, which should make non-GCC compilers happier too.

Thanks, because I am on LLVM.

> I'm a bit confused -- are you attempting to use picolibc in a Linux hosted environment, not building it using an embedded cross compiler?

No, I try to build binary for ARM Cortex-M3 and in the future I want to use other MCUs as well. 
Why did you think so?

keith-packard, 2020-05-04:
> No, I try to build binary for ARM Cortex-M3 and in the future I want to use other MCUs as well.
> Why did you think so?

Ah, you hadn't mentioned what your cross compilation configuration file looked like an I (incorrectly) assumed you weren't using one at all. Sorry for the confusion.

By the way, if you do happen to have a few POSIX OS functions available (open, close, read, write, lseek), then you can use the available tinystdio fopen/fdopen support to access your underlying file system.

I just tried version 1.4.1, using the supplied 'do-arm-configure' script (which uses GCC instead of clang). That doesn't build anything from the newlib/libc/reent directory at all (as I had hoped). Master has a number of fixes to libm corner cases and a few build cleanups, otherwise 1.4.1 is pretty current.

denizzzka, 2020-05-04:
Looks like latest ~master not contains this issue. Thank you!

> By the way, if you do happen to have a few POSIX OS functions available (open, close, read, write, lseek), then you can use the available tinystdio fopen/fdopen support to access your underlying file system.

No, I have only FreeRTOS and HAL library: https://github.com/denizzzka/d_c_arm_test/

closed 2020-05-04 by denizzzka