picolibc/picolibc #37: undefined symbols strtold and fflush

denizzzka, 2020-05-05:
My Meson config:
```
picolibc_sub = subproject(
    'picolibc',
    default_options: [
        'multilib=false',
        'newlib-io-c99-formats=true',
        'newlib-io-long-long=true',
        'newlib-io-long-double=true',
    ]
)
```

Error messages:
```
ld.lld: error: undefined symbol: fflush
ld.lld: error: undefined symbol: strtold
```

Generated picolibc.h says what defines are setted up correctly.

This looks like same issue as catched by @thmxv here: https://github.com/keith-packard/picolibc/issues/34#issuecomment-612561459

denizzzka, 2020-05-05:
upd: I removed complain about _WANT_IO_C99_FORMATS warnings: it was because also used `-D_WANT_IO_C99_FORMATS` compiler command line option

denizzzka, 2020-05-05:
I apologize for so few details. In general this is just same code as in the previous issue: https://github.com/keith-packard/picolibc/issues/36

keith-packard, 2020-05-05:
Ok, there's a bit to unpack here.

First off, there are a bunch of configuration options that apply to only the original newlib stdio code, and a few that apply only to the picolibc tinystdio code. The ones only supported by the original newlib stdio code are:

 * newlib-io-pos-args
 * newlib-io-c99-formats
 * newlib-io-long-double
 * newlib-io-float (tinystdio uses an application build-time option for this)
 * newlib-global-stdio-streams (tinystdio always uses global stdio streams)
 * newlib-fvwrite-in-streamio
 * newlib-fseek-optimization (tinystdio doesn't have a working fseek)
 * newlib-nano-formatted-io (same as io-float above)
 * newlib-wide-orient (I don't know what this does)

The newlib-io-long-double option (in current master) also attempts to control adding strtold to the library, but that code doesn't compile due to TLS changes which didn't get added to those functions. And that's why you don't have strtold. (note that on ARM, long double is the same as double, so strtold is the same as strtod). I'm working up patches to enable the non-io long double library functions (strtold, strtold_l, wcstold and wcstold_l) whenever the compiler has long doubles (even if those are just doubles with a different name).

fflush is a static inline function in current master; it should be a real function instead, so I'm changing that.

I've got things working, but it's 1am and I want to test things more and make decent commits before I push. Thanks again for your helpful testing!

denizzzka, 2020-05-05:
Thank you for a such good project!

johandc, 2020-05-05:
Also I am happy to see another user using the library as an embedded source. With mesons awesome subproject capability and build speed. Plus having the c library in the source tree leads to more stable and deterministic builds.

keith-packard, 2020-05-06:
Thanks to both of you for your kind comments! It's nice to be working on a problem that has vexed me since I started doing embedded ARM development over a decade ago. Suggestions on how to make this project useful to more people are great!

keith-packard, 2020-05-07:
Ok, getting 'strtold' working took a bit more time than expected -- RISC-V has 128-bit 'long double' values, and newlib's strtold only supported the intel 80-bit format. I ended up creating a stand-alone strtold for tinystdio (parallels the existing tinystdio strtod function) that supports both 80- and 128- bit long double types. For ARM, where 'long double' is just another name for 'double', strtold is now an alias for strtod, so it will work there too. I think this fixes all of the issues you've discovered here, so I'm going to close the bug. Please re-open if you discover things here that are still broken for you!

closed 2020-05-07 by keith-packard

denizzzka, 2020-05-07:
It works, many thanks!