picolibc/picolibc #38: _impure_ptr - gettimeofday

mp-17, 2020-05-05:
Hello, and thank you in advance for your time.

I am trying to compile Opus audio codec for a RISC-V target (Here the source https://github.com/xiph/opus).

First, I compiled with a RISCV-GCC release with _Newlib-nano_, and everything worked. 
Then I tried to compile using _picolibc_, and the linker threw an undefined reference to **_impure_ptr**. 
I defined **_impure_ptr** in the linker script to end the compilation, but the linker threw another undefined reference to **gettimeofday**.

I saw that these two symbols are defined in the _picolibc_ fork by SiFive https://github.com/sifive/picolibc.

For the compilation, I am using _picolibc_ with semihosting support.

Thank you for your help,
Matteo

keith-packard, 2020-05-06:
gettimeofday is not something picolibc can provide, as that requires information from the OS or other application environment. picolibc uses gettimeofday when you call time(), so that's presumably why this showed up as undefined.

I'm a bit confused as to how _impure_ptr could have been missing; that's defined in the library and should be available, unless there's some odd combination of build flags?

You shouldn't ever need to mix in symbols from newlib-nano with picolibc -- those are both providing the same layer (libc) on top of your operating system.

Oh, and sifive's picolibc is the same as upstream picolibc (if a bit delayed) -- I'm responsible for both :-)

mp-17, 2020-05-06:
>gettimeofday is not something picolibc can provide, as that requires information from the OS or other application environment. picolibc uses gettimeofday when you call time(), so that's presumably why this showed up as undefined. 

The strange fact is that with Newlib-nano everything compiled fine without any _undefined reference_.

> unless there's some odd combination of build flags?

I built picolibc with only "_-msave-restore_", and I am compiling OPUS targeting _rv32imc_ with "_-Os_, _-msave-restore_, _-ffunction-sections_, _-fdata-sections_ and _-gc-section_" enabled.

keith-packard, 2020-05-06:
I suspect you're ending up with something from newlib-nano getting linked in that references _impure_ptr -- there's nothing in picolib which uses _impure_ptr in the default configuration -- only the original newlib stdio still has references to that.

To make sure you're not getting any newlib bits, you can add -nostdlib to your link line, as long as you also add -lgcc to get the gcc library added. That should let us figure out what you're ending up using.

By the way, we (sifive) will be distributing a picolibc overlay for freedom-tools (if you're using that), and I've uploaded the risc-v toolchain to debian unstable which is equivalent to freedom-tools, and have uploaded picolibc, but that is stuck in the 'new' queue. All of that means that in the near future, you should be able to use pre-built verisons of all of this, which is known to work with freedom-metal and freedom-e-sdk on the platforms those tools support. The only catch is that these don't include rv32imc, only rv32imac. I can actually go add rv32imc to all of these, but I'd kinda like to know what target you've got that doesn't have atomic instruction support (the 'a' extension).

mp-17, 2020-05-10:
Thank you very much,
you are right, something was taken from Newlib. I don't understand why, as picolibc.specs updates the directives in such a way that the pico-libs are selected.
Unluckily, it now complains about a ___malloc_hook_ undeclared.  

Thank you for the tool, I am not currently using it, I used to build the toolchain by myself for custom multi-lib support. For the processor, I am targeting the CV32E40P core (rv32imc only, with an optional floating-point unit): https://github.com/openhwgroup/cv32e40p

keith-packard, 2020-05-10:
If you have a newer gcc, you can use -nolibc to skip newlib, otherwise it will be added after picolibc. With an older gcc, you need to use -nostdlib and -lgcc. That *should* keep newlib out of the link path.

__malloc_hook seems to be part of newlib's linux support, which seems a long ways from what you want.

Thanks for the pointer to that project; I will go suggest adding imc and imfc to the list of options supported by freedom-tools so that the default build of picolibc on debian includes those.

keith-packard, 2020-05-12:
The SiFive freedom-tools release, along with the Debian RISC-V GCC package both support this architecture already, so I think the remaining portion of this issue has been addressed. If you discover more issues, please re-open this, or create a new issue! Thanks for your contribution to picolibc.

closed 2020-05-12 by keith-packard

begriffs, 2024-08-15:
I'm also running into the linker error about `_impure_ptr` when I attempt to use semihosting. I'm wondering how a system newlib could be getting mixed up in my build.

I'm building with `-nostdlib` in my LDFLAGS, and manually specified `-lc -lgcc -lsemihost` to control the order of the libraries I link. The linker error comes from a part of my code that calls `fflush()`. Using verbose linker output, I see the following libraries selected, along with the symbols used from each:

* /usr/local/picolibc/arm-none-eabi/lib/thumb/v7e-m+fp/hard/libc.a
    * memcpy.c.o
    * libc_string_strrchr.c.o
    * libc_picolib_machine_arm_interrupt.c.o
    * libc_tinystdio_fflush.c.o
    * libc_tinystdio_snprintf.c.o
    * libc_tinystdio_vfprintf.c.o
    * libc_tinystdio_vprintf.c.o
    * libc_tinystdio_dtoa_ryu.c.o
    * libc_tinystdio_ryu_log10.c.o
    * libc_tinystdio_ryu_pow5bits.c.o
    * libc_tinystdio_ryu_table.c.o
    * libc_tinystdio_ryu_umul128.c.o
    * memset.c.o
    * libc_string_strchr.c.o
    * libc_string_strnlen.c.o
    * libc_tinystdio_dtox_engine.c.o
    * libc_tinystdio_filestrput.c.o
* /usr/local/lib/gcc/arm-none-eabi/12.2.0/thumb/v7e-m+fp/hard/libgcc/libgcc.a
    * _aeabi_uldivmod.o
    * _udivmoddi4.o
    * _dvmd_tls.o
* /usr/local/picolibc/arm-none-eabi/lib/thumb/v7e-m+fp/hard/libsemihost.a
    * iob.c.o
    * sys_getc.c.o
    * sys_putc.c.o
    * machine_arm_semihost-arm.S.o

keith-packard, 2024-08-15:
You'll either need to repeat -lc after -lsemihost or wrap `--start-group` and `--end-group` around `-lc -lsemihost` (and maybe include `-lgcc` in that as well). The semihost library depends on symbols from libc, so if you don't have libc.a after libsemihost.a, you'll end up getting undefined symbols. And it seems that `-nostdlib` isn't working right so that's pulling in newlib?

begriffs, 2024-08-17:
Thanks for the quick response.

I found my problem: I added `--specs=picolibc.specs` in my LDFLAGS, but I needed it in my CFLAGS as well. My code was pulling in newlib headers rather than picolibc headers, and newlib stdio.h references `_REENT` aka `_impure_ptr`.