picolibc/picolibc #39: clang -flto=full|thin causes strlen-thumb2-Os.S error

denizzzka, 2020-05-12:
Hi!

I try to compile with LTO by clang version 11:

```
clang -Isubprojects/picolibc/newlib/libc/machine/arm/293fc9c@@machine@sta -Isubprojects/picolibc/newlib/libc/machine/arm -I../subprojects/picolibc/newlib/libc/machine/arm -Isubprojects/picolibc/newlib/libc/tinystdio -I../subprojects/picolibc/newlib/libc/tinystdio -Isubprojects/picolibc -I../subprojects/picolibc -Isubprojects/picolibc/newlib/libc/include -I../subprojects/picolibc/newlib/libc/include -Xclang -fcolor-diagnostics -pipe -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -g -g -ffunction-sections -ftls-model=local-exec -flto=full -target thumbv7m-unknown-none-eabi -Oz -fshort-enums -fno-builtin -MD -MQ 'subprojects/picolibc/newlib/libc/machine/arm/293fc9c@@machine@sta/strlen.S.o' -MF 'subprojects/picolibc/newlib/libc/machine/arm/293fc9c@@machine@sta/strlen.S.o.d' -o 'subprojects/picolibc/newlib/libc/machine/arm/293fc9c@@machine@sta/strlen.S.o' -c ../subprojects/picolibc/newlib/libc/machine/arm/strlen.S
../subprojects/picolibc/newlib/libc/machine/arm/strlen-thumb2-Os.S:48:21: error: too many operands for instruction
1: ldrb.w r2, [r3], #1
                    ^
```

If line strlen.S: #include "strlen-thumb2-Os.S" is commented out (strlen-stub.c implementation used) compilation will be sucessful.

This is known LLVM issue: https://www.mail-archive.com/llvm-bugs@lists.llvm.org/msg37106.html

denizzzka, 2020-05-12:
Yes, for some reason this problem does not occur without -flto...

keith-packard, 2020-05-12:
Ok, I've reviewed the ARMv7-M architecture reference manual and I think the existing code is invalid ARM assembly. LDRB.W is supposed to force encoding T2 and can only look like this:
```
LDRB.W <Rt>,[<Rn>{,#<imm12>}]
```
This means the only valid addressing mode for LDRB.W is offset, while the desired instruction is post-indexed:
```
LDRB <Rt>,[<Rn>],#+/-<imm8>
```
This is only legal in encoding T3, which does not allow the .W suffix. It looks like GAS simply ignores the suffix as both T2 and T3 are 32-bit instructions. I examined the gas output from both versions of the code (with and without .W) and they generate identical binary instruction encodings (f893 2001), which is encoding T3.

I've pushed the two character fix which makes the file build under clang, and will submit it upstream to newlib.

closed 2020-05-12 by keith-packard