picolibc/picolibc #42: __aeabi_read_tp: Clang wants to use different places for return value

denizzzka, 2020-05-28:
For [this](https://github.com/keith-packard/picolibc/blob/38236f58d32d7be47882944967e146f59f03f4d9/newlib/libc/picolib/machine/arm/tls.c#L52) function:
```C
void *
__attribute__((naked))
__aeabi_read_tp(void)
{
	/* Load the address of __tls */
	asm("ldr r0,1f");
	/* Dereference to get the value of __tls */
	asm("ldr r0,[r0]");
	/* All done, return to caller */
	asm("bx lr");
	/* Holds the address of __tls */
	asm(".align 2");
	asm("1: .word __tls");
}
```

Clang with "-O0" option wants to get result value from stack instead of r0 register. If "-O3" or "-Oz" used it wants to obtain result by another different ways, but also not from r0. As result of attempt to use this function binary obtains (silently!) wrong value if compiled with Clang.

It is possible to use some "clobber" or something else to point where is result placed for compiler?

denizzzka, 2020-05-28:
Perhaps this is not a bug and I should write my own such function? But this call is hardcoded into ldc2, and also binary can be compiled from C and D sources both, and both of them need to call this function.

keith-packard, 2020-05-28:
No, definitely a picolibc bug. I've moved the function to a .S file instead of attempting to have the C compiler generate the code we want. I'll push that out once some (unrelated) tests are complete. Thanks for the bug report!

denizzzka, 2020-05-29:
Ok, thanks

I temporary changed my copy of tls.c by this way and it works:
```C
/* This needs to be global so that __aeabi_read_tp can
 * refer to it in an asm statement
 */
size_t *__tls;

/*
 * This cannot be a C ABI function as the compiler assumes that it
 * does not modify anything other than r0 and lr. So we create a
 * 'naked' function that respects those limitations.
 */
void *
__aeabi_read_tp(void)
{
	return (void*) *__tls;
}

/* The size of the thread control block.
 * TLS relocations are generated relative to
 * a location this far *before* the first thread
 * variable (!)
 */
#define TCB_SIZE	8

void
_set_tls(void *tls)
{
	__tls = (size_t *) tls - TCB_SIZE;
}

```

keith-packard, 2020-05-29:
Yeah, that might well work with clang. With GCC, at -Os optimization, it used another register and broke code calling it.

In any case, I've pushed out a version written in assembly, so that should work on clang as well as it does with gcc.

denizzzka, 2020-05-30:
> I've pushed out a version written in assembly, so that should work on clang as well as it does with gcc

I tested it with `ldc2` (not clang!) without succes: it generates same code as previous version: __aeabi_read_tp puts result into r0, and then tries to get result from stack:

[image]

denizzzka, 2020-05-30:
>  I've pushed out a version written in assembly, so that should work on clang as well as it does with gcc.

I tried this commit without success, it fails with same symptoms.

Clang compiles C version of __aeabi_read_tp (posted previously):
```C
void * __aeabi_read_tp(void)
{
	return (void*) *__tls;
}
```
into:
[image]

Generated by LLVM caller code is equal for my and your versions of __aeabi_read_tp:
[image]

And it works with C version, but not with asm version.

denizzzka, 2020-05-30:
I forgot to call _set_tls before use __aeabi_read_tp.
I am not sure, but may be this causes this problem, not asm issues.
I will reopen this issue afterch additional check.

closed 2020-05-30 by denizzzka

denizzzka, 2020-06-02:
Current variant tested and it works ok.
Problem was in https://github.com/keith-packard/picolibc/issues/45 and GC configuration on my side.