picolibc/picolibc #45: sbrk() returns wrong value?

denizzzka, 2020-06-02:
This is about `sbrk()` placed in file
https://github.com/keith-packard/picolibc/blob/a5c96342f91edd5ee9357a8a9c02dc4b6f6daaa3/newlib/libc/picolib/picosbrk.c#L46

I faced with situation described in these lines inside of `sbrk()` body:
```C
	if (brk + incr > __heap_end)
		return NULL;
```

sbrk() was called by `sbrk_aligned()` from https://github.com/keith-packard/picolibc/blob/58559a42c49891bca33d363b3ed781bd6909f991/newlib/libc/stdlib/nano-mallocr.c#L191

Inside of `sbrk_aligned()` NULL is not treated as error:
```C
        /* p is not aligned, ask for a few more bytes so that we have s
         * bytes reserved from align_p. */
        p = sbrk(align_p - p);
        if (p == (void *)-1)
            return p;
```

Probably, sbrk() should return -1 instead of NULL in described case.

denizzzka, 2020-06-02:
`return -1;` instead of `NULL` works without problems on my side

keith-packard, 2020-06-02:
Yup, thanks for the bug report. Pushed the obvious fix to master.

closed 2020-06-02 by keith-packard