picolibc/picolibc #46: _set_tls: proposal to remove TCB_SIZE

denizzzka, 2020-06-04:
File `newlib/libc/picolib/machine/arm/tls.c` contains line:
```C
__tls = (uint8_t *) tls - TCB_SIZE;
```
What about remove of TCB_SIZE?

Rationale:
* neighboring `riscv/tls.c` is not contains such constant
* TCB can be different sized on different OSes

This is not very important, I always can add desired value when calling, but nonetheless.

denizzzka, 2020-06-04:
Oh, I didn’t figure it out, this is a hardware feature! Incorrect issue, closing

closed 2020-06-04 by denizzzka

denizzzka, 2020-06-04:
In fact, I did not understand how this works! :-)
If I remove this `- TCB_SIZE` then my program fails at some point.

keith-packard, 2020-06-04:
Yeah, GCC has a built-in notion of this value, which makes it effectively a part of the ABI.

denizzzka, 2020-06-04:
Oh! I am on Clang - looks like LLVM have same built-in!

Thanks!