picolibc/picolibc #48: document meson and ninja dependency

ThomasHornschuh, 2020-06-06:
I did not find any hint about the required version of ninja and meson. I'm still using quite an old Ubuntu 16.04 LTS on one of my RISC-V development VMs, the version of ninja and meson provided by apt are really outdated.

Installing meson with sudo pip3 install meson gives 0.54.2 currently, this seems new enough.
I built ninja from sources.

At least some hints to the required versions should be added to the readme.

keith-packard, 2020-06-06:
That's a great suggestion. I've added a couple of paragraphs to the top of the doc/build.md document describing the version requirements and providing some hints about how to get new enough bits.

closed 2020-06-06 by keith-packard

ThomasHornschuh, 2020-06-07:
Wow, that was fast :-)

keith-packard, 2020-06-07:
You're most welcome!