picolibc/picolibc #50: Can not build native picolib for testing

archi, 2020-07-22:
Hi, I am currently evaluating libc's for use with CompCert (our targets are various arm, aarch64, various riscv and powerpc).
However, I'm already struggling to build native executable with GCC 10.1.10 :'(

I followed the [doc/testing.md](https://github.com/keith-packard/picolibc/blob/main/doc/testing.md) and did:

> mkdir build.native/out && cd build.native
> ../do-native-configure --prefix=$(realpath out) -Dspecsdir=$(realpath out) > [meson.log](https://github.com/keith-packard/picolibc/files/4960349/meson.log)
> ninja -v > [ninja.log](https://github.com/keith-packard/picolibc/files/4960348/ninja.log)
> ninja install

This doesn't cause any errors.
But if I now try to compile a simple test

> echo "int main (void) { return 0; }" > test.c
> gcc --specs out/picolibc.specs test.c

All I get is a

> /usr/bin/ld: /local/hdd/meyer/git/picolibc/build.native/out/lib/picolibc/lib/../lib/crt0.o: in function \`__start':
> /local/hdd/meyer/git/picolibc/build.native/../picocrt/shared/../crt0.h:84: undefined reference to \`_set_tls'
> collect2: error: ld returned 1 exit status

Is building native executables not supported? Or am I doing something wrong?

keith-packard, 2020-07-22:
picolibc is not designed to be used on top of a full POSIX environment, it's designed to be used on embedded systems. The 'native' configuration is used when testing the library as it relies on the host C library for access to system calls. Because of that, you need to use the native C library startup code instead of the picolibc one, which means you don't want to use picolibc.specs.

Or perhaps you're doing something I don't quite understand?

archi, 2020-07-23:
Thanks Keith, you understood right. We're currently running our x86 (both 64 and 32 bit) CI tests natively on Linux, which I would like to keep. I naively assumed that a _do-native-configure_-picolib would result in a picolib that would, well, "just run" natively as well ;) Thank you for clearing that up.
So my "best option" for running x86_64 natively probably is to copy&adapt the compile&link commands from ninja?

By the way, I did try aarch64-linux-gnu. The library builds and executables can be run in qemu (using the included run-aarch64). Though [some tests are failing for aarch64-linux-gnu on 1.4.4-1](https://github.com/keith-packard/picolibc/files/4964634/btw-aarch64-linux-gnu-errors.log), but it's less with branch main. I suppose you're aware of this?

keith-packard, 2020-07-23:
> Thanks Keith, you understood right. We're currently running our x86 (both 64 and 32 bit) CI tests natively on Linux, which I would like to keep. I naively assumed that a [...]

I supposed that could be made to work by creating a special picolibc.specs that linked against both picolibc and the host libc. The test.specs file used to build tests within picolibc might be some help if you want to experiment. Being able to use picolibc on the linux host for further application testing before deploying to an embedded host might be useful?

> By the way, I did try aarch64-linux-gnu. The library builds and executables can be run in qemu (using the included run-aarch64). Though [some tests are failing for aarch64-linux-gnu on [...]

aarch64 testing support, including semihosting, crt0, and TLS setup was added in 1.4.5. You can see the results of running those tests in the current picolibc CI results: https://github.com/keith-packard/picolibc/runs/862266941?check_suite_focus=true  You should be able to reproduce those results if you like.

keith-packard, 2020-07-28:
I've done some experimentation with this and adjusted the picolibc build system so that you can install and use picolibc on a native POSIX system for testing purposes. Check out [os.md](https://github.com/keith-packard/picolibc/blob/main/doc/os.md#building-picolibc-on-native-posix-systems) which describes how this works. Note that the provided do-native-configure is all set up for this. I'm going to mark this issue closed now; please re-open if you have additional issues.

closed 2020-07-28 by keith-packard

archi, 2020-08-13:
Thanks Keith, I can confirm this also works here (I only added a custom --prefix to avoid installing the lib globally).