picolibc/picolibc #70: Split crt0.c in multiple .s files

archi, 2020-09-08:
As we discussed on my fork, you planned to adapt the picocrt build such that the assembly code resides in `.s`-files as opposed to `.c` with `asm`-statements.

Do you want adapt the build system for this, or should I take a look at it?

keith-packard, 2020-09-11:
I'd be happy to review this if you can manage; been a bit busy 'round here in Oregon this week...

archi, 2020-09-14:
> I'd be happy to review this if you can manage

Okay, we've an upcoming release, so this isn't my top priority; but I'll take a look.

> been a bit busy 'round here in Oregon this week...

Uff, I assumed you were occupied with more pleasant things. I hope the situation in Oregon improves soon.

archi, 2020-11-23:
I wrote a `crt0.S` that should be a 1:1 replacement for `picocrt/machine/riscv/crt0.c`. I looked at what gcc & CompCert produce and adapted it. It is essentially a assembler variant of `picocrt/machine/riscv/crt0.c` with `__start` (from `picocrt/crt0.h`) already inlined. Not sure if that's exactly what you had in mind, but obviously nothing is set in stone.

Good things:
- Uses no stack at all
- Seems to build & test successfully for my riscv32 config with both gcc and CompCert (let the CI check for the rest)
- Integration into `meson.build` is as simple as a few `s/crt0.c/crt0.S/`

Things I am not sure about:
- Should `__start` really be inlined? Maybe better call it, so hypothetical future changes also affect the `.S` variant (obviously this doesn't work with `__start` living in `crt0.h`)
- What to do at the end of `_cstart`? For now I just `j _start`.

I'll prepare a small PR with the necessary changes to the build files. [Here is the preliminary variant](https://github.com/picolibc/picolibc/files/5583305/crt0.txt) (as `.txt`, because GitHub doesn't like `.S`...)

archi, 2020-11-25:
With #100 merged, RISC-V is now done. I added a fancy list to the initial post.

I am not sure if I get around porting the other two as well any time soon. If I start on them, I'll add a comment here.
**If anyone else steps up: Feel free**, just say something here so we don't end up doing double the work :)

closed 2026-11-10 by keith-packard