picolibc/picolibc #75: qemu-system-riscv32 does not support semihostconfig

elliott-wen, 2020-09-14:
Hi, when I tried to run qemu-system-riscv32 to simulate the hello-world program, I got the following error

qemu-system-riscv32: -semihosting-config chardev=stdio0: Option not supported for this target

At the very beginning, I though it's because of my old qemu version. But after trying the latest qemu 5.1 release, I still got the same error. Do you apply some magic patches on your qemu?

Cheers

archi, 2020-09-14:
Hi @elliott-wen, try using Keith's `riscv-semihost` qemu branch at https://github.com/keith-packard/qemu/tree/riscv-semihost - this worked for me.

elliott-wen, 2020-09-14:
thanks for pointing that out.
 Cheers

keith-packard, 2020-09-14:
I've been asking for this to be merged upstream, but so far it hasn't happened.

closed 2020-09-14 by keith-packard