picolibc/picolibc #82: strlen is removed for ARM Cortex M3?

denizzzka, 2020-10-02:
```
ld.lld: error: undefined symbol: strlen
>>> referenced by exception.d:585
>>>               firmware.elf.p/subprojects_druntime_src_core_exception.d.o:(_d_assertp)
>>> referenced by exception.d:603
>>>               firmware.elf.p/subprojects_druntime_src_core_exception.d.o:(_d_unittestp)
>>> referenced by exception.d:621
>>>               firmware.elf.p/subprojects_druntime_src_core_exception.d.o:(_d_arrayboundsp)
>>> referenced 4 more times
>>> did you mean: strnlen
>>> defined in: subprojects/picolibc/newlib/libc/libc.a(strnlen.c.o)
clang: error: ld.lld command failed with exit code 1 (use -v to see invocation)
ninja: build stopped: subcommand failed.
```

```meson
picolibc_sub = subproject(
    'picolibc',
    default_options: [
        'multilib=false',
        'newlib-io-c99-formats=true',
        'newlib-long-time_t=true',
        'newlib-multithread=true',
    ]
)
```

Bisect says that blame is on 0e7b776ae58951a3152fc35709497edd454188b5.

denizzzka, 2020-10-02:
`nm subprojects/picolibc/newlib/libc/machine/arm/libmachine.a.p/strlen.S.o` says that no one symbols are defined

denizzzka, 2020-10-02:
Cross file:
```ini
[host_machine]
system = 'bare metal'
cpu_family = 'arm'
cpu = 'cortex-m3'
endian = 'little'

[binaries]
d = 'ldc2'
c = 'clang'
cpp = 'clang'
ar = 'llvm-ar'
strip = 'llvm-strip'

[properties]
d_args = [
        '-release',
        '-O0',
        '--checkaction=D',
        '--mtriple=thumbv7m-unknown-none-eabi',
        '--unittest',
        '--d-version=CoreUnittest',
#        '--d-debug=PRINTF',
#        '--d-debug=MARK_PRINTF',
#        '--d-debug=LOGGING',
#        '--d-debug=qsort',
        '--d-debug=SENTINEL',
        '--d-debug=INVARIANT',
    ]
c_args = [
        '-target', 'thumbv7m-unknown-none-eabi',
        '-O0',
        '-fshort-enums', #libopencm3 compatibility
    ]
cpp_args = [
        '-target', 'thumbv7m-unknown-none-eabi',
        '-O0',
        '-fno-rtti', #prevents undefined symbol: vtable for __cxxabiv1::__si_class_type_info
        '-fno-exceptions', #prevents undefined reference to `__gxx_personality_v0'
        '-funwind-tables', #for correct calling of libunwind _Unwind_RaiseException
        '-fshort-enums', #libopencm3 compatibility
    ]
d_link_args = [
        '--mtriple=thumbv7m-unknown-none-eabi',
        '-L=--static', #prevents looking for __gxx_personality_v0
    ]
c_link_args = [
        '-target', 'thumbv7m-unknown-none-eabi',
        '--no-standard-libraries', #TODO: move to meson
        '-z', 'norelro', #prevents ld.lld: error: section: .tdata is not contiguous with other relro sections
    ]
cpp_link_args = [
        '-target', 'thumbv7m-unknown-none-eabi',
        '--no-standard-libraries', #TODO: move to meson
    ]

#c_stdlib = ['picolibc', 'picolibc_dep'] #broken: https://github.com/mesonbuild/meson/issues/1553
```

keith-packard, 2020-10-03:
Nice catch. Looks like I was relying on having picolibc.h get included everywhere PREFER_SIZE_OVER_SPEED was being used, which is not the case in a bunch of files. Instead of attempting to find and fix every place to make sure picolibc.h is included, I'm adding -include picolibc.h to the compiler flags to force those symbols to always be available. Newlib places some of these on the command line in the autotools build, which I'd prefer to avoid. This works for me using your cross file, making strlen get defined from strlen.S

I looked at how painful this would be to fix by making sure every file included picolibc.h somehow and decided that just forcing the issue on the command line would be far easier and more reliable.

I am rather surprised that nothing in the test suite uncovered this problem because it occurs under the default gcc builds too.

I'll close this once the CI system shows green; the proposed fix is on main at this point.

closed 2020-10-04 by keith-packard