picolibc/picolibc #87: e_log10.c is only for 32bits?

denizzzka, 2020-10-19:
Hi!

Found what  base 10 logarithm calculation is preceded by: https://github.com/picolibc/picolibc/blob/034511369a59d3cfd48eb83d7d4f0e759de74d20/newlib/libm/math/e_log10.c#L49):

But looks like actually it is 32bit function.

I tried to call log10(1000000) with 8 byte double value and this function returns negative value: -301.65265204006647 instead of 6.

denizzzka, 2020-10-19:
although, a million is not much for 32 bit

keith-packard, 2020-10-19:
On some targets, 'double' is the same as 'float', and both are 32-bit values (I know 8-bit AVRs do this, for one example). For those platforms, picolibc creates aliases so that math functions on doubles end up calling the same code as on floats. That means the 64-bit double functions are only compiled when doubles aren't also 32-bits. You're seeing one of the guard #ifndef values around this function:
```
#ifndef _DOUBLE_IS_32BITS
...
#endif  /* defined(_DOUBLE_IS_32BITS) */
```
Note that the first line uses #ifndef, so this code is compiled whenever `DOUBLE_IS_32BITS` is *not* defined.

A simple test on 32-bit arm, 64-bit arm and 32-bit RISC-V using the default picolibc configuration shows this function working as expected:
```
#include <stdio.h>
#include <math.h>

volatile double million = 1000000.0;

int
main(void)
{
	printf("log10(%17g) = %17g\n", million, log10(million));
	return 0;
}
```
I had to use a global volatile variable to avoid having the compiler optimize away the `log10` call at compile time. When run on any of these architectures, I get:
```
log10(            1e+06) =                 6
```
If you've got a case that fails, please feel free to add a custom test for it using the existing test infrastructure; creating a PR with that will have it run on all of the CI configurations which may uncover which configuration options are causing this call to do the wrong thing.

denizzzka, 2020-10-20:
Found what ANY argument passed to this function calculated wrong.
I use clang, maybe problem with it? log10 implementation contains so much number of binary shifts, intuition says the problem is out there.

denizzzka, 2020-10-20:
Got it!

__IEEE_BIG_ENDIAN is defined for cortex m3 targed and EXTRACT_WORDS macro from fdlibm.h confuses words!

denizzzka, 2020-10-20:
Working dirty fix: add this line into ieeefp.h instead of ARM defines block:
```C
#define __IEEE_LITTLE_ENDIAN
```

keith-packard, 2020-10-20:
wow! good catch. I've pushed a patch that should work for you t- the arm-clang-fp-test branch; I discovered that clang doesn't set __VFP_FP__ when doing soft floats, but it does set __SOFTFP__. Assuming that soft floats always use the native byte order, I've added a check that looks at that in addition to __VFP_FP__. That makes clang happy in both hard float and soft float configurations at least.

denizzzka, 2020-10-20:
CI for clang doesn't covers this code?

denizzzka, 2020-10-21:
>  I've pushed a patch that should work for you to the arm-clang-fp-test branch

Tested, works for me, thanks!

keith-packard, 2020-10-21:
I'm testing CI for clang on ARM -- I found a couple of pretty important bugs which I've fixed in nano-malloc and libm (the fma asm code was busted). It's now passing the full CI set here on my machine; I've added CI runs using clang to the github actions which should keep things working in the future.

keith-packard, 2020-10-21:
CI has also discovered that clang mis-compiles 'memalign' from the original newlib malloc code. I don't have a fix for that, so I've kludged around it by disabling the memalign tests for now. I'd suggest using 'nano-malloc' (which is the default) as that gets more fully tested.

keith-packard, 2020-10-21:
And everything is building, but I need to update the qemu-system-arm version I'm using for testing as upstream can't test thumbv7e+fp. Will be a couple of days before I get back to this. Tests are passing on thumbv7m.

keith-packard, 2020-10-25:
Ok, I had to re-build qemu for debian testing (picolibc uses a custom build to provide the necessary test coverage). With that, all of the clang tests are building and passing on both v7m and v7e+fp, so I think we're done here. The best part is that any future breakage will be caught by CI before being released.

closed 2020-10-25 by keith-packard