picolibc/picolibc #92: Proposal about removing multilib support

denizzzka, 2020-10-20:
If I correct understand, multilib is way to build picolibc binaries for many architectures by one meson build call.

This behavior is against very idea of ​​a Meson. And this leads to more complex code of meson.build files what should reimplement some functionality what already implemented inside of Meson.

I propose to remove this functionality and/or replace it by (generated) cross files for integration testing purposes.

keith-packard, 2020-10-20:
While multilib support does make the meson.build files a bit more complicated, it speeds up building the RISC-V and ARM versions of the library when building toolchains and testing a great deal, and it makes installing those cleaner as the common files are easily installed only once. That's a pretty big piece of where picolibc is used. I could see disabling multilib support by default though; that would make simple uses of the library, like embedding it into another system, easier to manage.

denizzzka, 2020-10-20:
Multilib also complicates second-level meson.build files constructions similar to:
```meson
foreach target : targets
  value = get_variable('target_' + target)
  libobjs = []

  foreach libname : libnames
    libobjs += get_variable('lib_' + libname + target).extract_all_objects(recursive:true)
  endforeach
[...]
```

gebart, 2020-10-20:
Could it be possible to hoist multilib out of meson to make it more of an external process if that simplifies the rest of the build config?
Something like "for each multilib, do meson setup, meson compile, meson install" with the proper prefixes and compiler flags injected in each iteration.

denizzzka, 2020-10-20:
meson.build can provide shell script generator for this purpose :-)
For example, it can parse cross files directory and make execute line for each found file

keith-packard, 2020-10-21:
That would make the multilib build go a *lot* slower as you'd have dozens of meson runs. It also complicates installing in a multilib environment as you need to ensure that only one copy of the shared files gets installed.  Doing this outside of meson would involve a raft of kludgy scripting, including generating cross-compilation control files using templates or some such. It's at least straightforward to add the necessary looping in each sub-directory to get meson to do everything for us, even if it's not as simple as a single-target build.

keith-packard, 2020-10-25:
I'm going to close this issue for now; given the range of requirements for picolibc, being able to quickly and reliably create multilib builds remains important to enough of us that I will leave it in place. Switching the default to single libs would be a reasonable change though, so if you want to propose a patch that does that, please feel free.

closed 2020-10-25 by keith-packard