picolibc/picolibc #94: exe_wrapper won't works

denizzzka, 2020-10-21:
If I try to use `exe_wrapper` in my own project what uses picolibc:
```ini
[host_machine]
system = 'bare metal'
cpu_family = 'arm'
cpu = 'cortex-m3'
endian = 'little'

[...]

exe_wrapper = [
        'qemu-system-arm',
                '-machine', 'mps2-an511',
                '-nographic',
                '-semihosting',
                '-kernel',
        ]
```

I start meson setup and immediately face with:
```
Build machine cpu family: x86_64
Build machine cpu: x86_64
Host machine cpu family: arm
Host machine cpu: cortex-m3
Target machine cpu family: arm
Target machine cpu: cortex-m3

|Executing subproject picolibc method meson 
|
|Project name: picolibc
|Project version: 1.4.6
|C compiler for the build machine: cc (gcc 9.3.0 "cc (Debian 9.3.0-18) 9.3.0")
|C linker for the build machine: cc ld.bfd 2.35.1
qemu: fatal: Lockup: can't escalate 3 to HardFault (current priority -1)

R00=00000000 R01=00000000 R02=00000000 R03=00000000
R04=00000000 R05=00000000 R06=00000000 R07=00000000
R08=00000000 R09=00000000 R10=00000000 R11=00000000
R12=00000000 R13=ffffffe0 R14=fffffff9 R15=00000000
XPSR=40000003 -Z-- A handler
FPSCR: 00000000
```

Log says what Meson tries to run executable `builddir/meson-private/sanitycheckc_cross.exe`, but, probably, exe_wrapper isn't supports memory layout of this binary - first bytes for 'mps2-an511' device must contain interrupts vector.

denizzzka, 2020-10-21:
picolibc is used by this meson.build code:
```meson
picolibc_sub = subproject(
    'picolibc',
    default_options: [
        'multilib=false',
        'newlib-io-c99-formats=true',
        'newlib-long-time_t=true',
        'newlib-multithread=true',
    ]
)
picolibc_dep = picolibc_sub.get_variable('picolibc_dep')
```
Then picolibc_dep is used in executable()

keith-packard, 2020-10-21:
I've pushed a fix for this to the arm-clang-fp-test branch -- the hack I had stuck in picocrt to pull in an interrupt vector doesn't work on clang, so I created a different hack. That branch has examples which build with clang and run under qemu.

denizzzka, 2020-10-21:
> I've pushed a fix for this to the arm-clang-fp-test branch

Issue is still here

denizzzka, 2020-10-21:
Found full log of sanity check compilation and run:
```
Executing subproject picolibc method meson 

Project name: picolibc
Project version: 1.4.6
None of 'CFLAGS' are defined in the environment, not changing global flags.
None of 'LDFLAGS' are defined in the environment, not changing global flags.
None of 'CPPFLAGS' are defined in the environment, not changing global flags.
None of 'CC_LD' are defined in the environment, not changing global flags.
Sanity testing C compiler: clang
Is cross compiler: True.
None of 'CC_LD' are defined in the environment, not changing global flags.
Sanity check compiler command line: clang /home/denizzz/Dev/d_c_arm_test/builddir/meson-private/sanitycheckc.c -o /home/denizzz/Dev/d_c_arm_test/builddir/meson-private/sanitycheckc_cross.exe -target thumbv7m-unknown-none-eabi -fshort-enums -pipe -D_FILE_OFFSET_BITS=64 --no-standard-libraries -z norelro
Sanity check compile stdout:

-----
Sanity check compile stderr:
ld.lld: warning: cannot find entry symbol _start; defaulting to 0x200E4

-----
Running test binary command: qemu-system-arm -machine mps2-an511 -nographic -semihosting -kernel /home/denizzz/Dev/d_c_arm_test/builddir/meson-private/sanitycheckc_cross.exe

subprojects/picolibc/meson.build:35:0: ERROR: Executables created by c compiler clang are not runnable.
```

denizzzka, 2020-10-21:
And if I provide my own linker script from current project (with reset_handler as entry point) it returns error:
```
Sanity check compile stderr:
ld.lld: warning: cannot find entry symbol reset_handler; defaulting to 0x21000000
```

So, looks like here is Meson design issue?

denizzzka, 2020-10-21:
It is possible just ignore sanity check:
```ini
[properties]
skip_sanity_check = true
```

So, this problem isn't actual.

closed 2020-10-21 by denizzzka