picolibc/picolibc #96: memory layout problem struct __file

rockcarry, 2020-11-04:
memory layout problem of struct:

struct __file {
	__ungetc_t unget;	/* ungetc() buffer */
	uint8_t	flags;		/* flags, see below */
#define __SRD	0x0001		/* OK to read */
#define __SWR	0x0002		/* OK to write */
#define __SERR	0x0004		/* found error */
#define __SEOF	0x0008		/* found EOF */
#define __SCLOSE 0x0010		/* struct is __file_close with close function */
	int	(*put)(char, struct __file *);	/* function to write one char to device */
	int	(*get)(struct __file *);	/* function to read one char from device */
	int	(*flush)(struct __file *);	/* function to flush output to device */
};

the sizeof __ungetc_t is incorrect, when I implement the stdin/stdout __iob interface.
it cause puts function call stdout->get, when I runing a helloworld test program:

#include <stdlib.h>
#include <stdio.h>

int main(void)
{
    printf("hello world !\n");
    return 0;
}

I tested and found __GCC_HAVE_SYNC_COMPARE_AND_SWAP_4 is not defined, when I compile my source code.
but why it defined when I using ninja to build picolibc ?

keith-packard, 2020-11-04:
That's odd -- this symbol is internally defined by GCC itself, not in a header file. Are you using a different compiler or different compiler options for your application than when building picolibc?

You can see what symbols are available from the compiler with this command:

 ```$ gcc -dM -E - < /dev/null | grep SYNC```

For example, to check what support my arm-none-eabi compiler has when building for the cortex-m3, I do:
```
$ arm-none-eabi-gcc -mcpu=cortex-m3 -dM -E - < /dev/null | grep SYNC
#define __GCC_HAVE_SYNC_COMPARE_AND_SWAP_1 1
#define __GCC_HAVE_SYNC_COMPARE_AND_SWAP_2 1
#define __GCC_HAVE_SYNC_COMPARE_AND_SWAP_4 1
$
```

We should probably fix this by having picolibc detect these symbols at build time and create a suitable picolibc-specific definition that could go in picolibc.h so that we won't need to use the same compiler to build the library and applications.

I've recently been working on better c++ support (including libstdc++), and that needs fixes here too as c++ can't use stdatomic.h and doesn't define `_Atomic` in the compiler. I'll take your issue into account while fixing that.

Thanks for your bug report!

keith-packard, 2020-11-04:
oh. this is trickier than it should be -- the definitions depend on the compiler options so we can't stick something in picolibc.h as that is only built once for all versions of the library. Maybe this should just be architecture-specific instead of checking a compiler definition so that it is always 32 bits on RISC-V (which never has 2-byte atomics) and 16 bits elsewhere.

keith-packard, 2020-11-04:
I've pushed a change that sets the type of this variable based solely on the base architecture so it should not vary depending on compiler options. That's not optimal for RISC-V variants without atomics, but those aren't terribly common.

closed 2020-11-04 by keith-packard

rockcarry, 2020-11-05:
yes I think so.

The root cause maybe I compile picolibc with the config rv32imac, cross-rv32imac.txt. but my toolchain only support the rv32imc.

keith-packard, 2020-11-05:
yeah, imc is not one of the multi-libs options configured into gcc by default, and without the 'a' extension, you won't have the atomics necessary. 

If you want to use your existing picolibc source, you can add `-Datomic-ungetc=false` to the meson configuration command. That will skip the atomic instructions and use a 2-byte type for `__ungetc_store_t`.