picolibc/picolibc #97: newlib-multithread=true isn't enables MALLOC_LOCK in nano-mallocr.c

denizzzka, 2020-11-11:
My config:
```meson
picolibc_sub = subproject(
    'picolibc',
    default_options: [
        'multilib=false',
        'newlib-global-errno=true',
        'newlib-io-c99-formats=true',
        'newlib-long-time_t=true',
        'newlib-multithread=true',
        'newlib-tinystdio=true',
        'thread-local-storage=false',
    ]
)
```

denizzzka, 2020-11-11:
Changing `newlib-tinystdio to` `false` causes:

```
fseeko64.c
../subprojects/picolibc/newlib/libc/stdio64/fseeko64.c:356:29: error: expected '}'
#endif /* __LARGE64_FILES */
                            ^
../subprojects/picolibc/newlib/libc/stdio64/fseeko64.c:92:1: note: to match this '{'
{
^
1 error generated.
```

I am on ~main, 12d5617f5c8f854c63b9435a8a9f0c306de1d96f

keith-packard, 2020-11-12:
So, I'm a bit confused by this report. In this configuration, malloc calls __malloc_lock and __malloc_unlock to perform locking, letting you re-define those to perform whatever target-specific operations are necessary. Let me know what you are trying to do and I'll see how that can work.

Your second note uncovers a bug in upstream newlib where the 64-bit specific versions of fseek are mis-using some file locking macros. I've pushed a fix for that, copied over from the regular fseek code.

denizzzka, 2020-11-12:
>  **letting you re-define those**

Got it!
Sorry, previously I didn't read about this and thought it is some built-in locks but for some reason they are not compiled-in.

Thanks for the `fseek` fix!

closed 2020-11-12 by denizzzka

keith-packard, 2020-11-12:
Pretty hard to do built-in locking when you don't control task scheduling. That's why I switched to using atomics for tinystdio; no locking necessary. Thanks for finding the fseek bug; looks like we need another configuration for CI to churn on so we don't break this again. I wonder if you can push changes to the .github directory and get those to actually run?

denizzzka, 2020-11-12:
> Pretty hard to do built-in locking when you don't control task scheduling.

In my project I have FreeRTOS and also 3rd-party spinlock, so it will be easy.

> I wonder if you can push changes to the .github directory and get those to actually run?

I will try

denizzzka, 2020-12-16:
@keith-packard 

>  I wonder if you can push changes to the .github directory and get those to actually run?

I remember I promised it, but currently I havent time for it, sorry. It is necessary to write down somewhere this into TODO for not to forget.

keith-packard, 2020-12-17:
Thanks for the reminder. I've added tests with newlib-multithread and newlib-retargetable-locking set, and also added locking to the posixio functions in tinystdio.