picolibc/picolibc #99: Value of __LDBL_MANT_DIG__

stuarch, 2020-11-16:
I found some of `if defined(_HAVE_LONG_DOUBLE)` come with `__LDBL_MANT_DIG__ == 64` like [here](https://github.com/picolibc/picolibc/search?q=__LDBL_MANT_DIG__), but my riscv64-elf-gcc always define `__LDBL_MANT_DIG__` as 113 and these functions just don't work.

I think `__LDBL_MANT_DIG__ == 64` is true when length of `long double` is 12 bytes, but the [document](https://riscv.org/wp-content/uploads/2015/01/riscv-calling.pdf) says the length of long double on risc-v should be 16 bytes.

Please help me check if this is correct, thanks.

keith-packard, 2020-11-16:
Yes, this is all correct -- the existing long double code works for 80-bit precision floats as found on Intel processors. It does not work on 128 bit floats as found on RISC-V processors, which is why I haven't spent much time worrying about it. Rewriting all of the math functions to work on 128 bit floating point numbers will be a significant amount of work. For now, because these functions will not work correctly, they do not get compiled at all.

closed 2020-11-16 by keith-packard

stuarch, 2020-11-16:
Ok I understood. Thanks a lot.