picolibc/picolibc #102: Tests for platforms without fenv.h

tdaede, 2020-11-29:
I'm working on adding build and test support for m68k, which doesn't have a newlib sys/fenv.h header (and therefore no FE exception support). This works on newlib because all newlib tests check for the presence of FE_INVALID etc. I was wondering if this is the approach you would like to take on picolibc as well before I plow ahead, with the caveat that it will result in a lot of ifdefs in math_errhandling_tests.c.

keith-packard, 2020-11-29:
I *think* everything except that one test program should build without new ifdefs. I should have written the new test to run on machines without fenv.h, but I was focused on getting the fenv stuff working correctly. Feel free to hack up that test to make it build for you.

keith-packard, 2021-06-05:
This should all work correctly now -- there's a default fenv.h which defines the few required macros (like FE_ALL_EXCEPT) to let applications know they aren't getting any useful fenv support.

closed 2021-06-05 by keith-packard