picolibc/picolibc #103: What is the relation of picolibc vs newlib-nano

fanghuaqi, 2020-12-04:
Hi guys, I want to learn about what is the difference of picolibc vs newlib-nano, is there any comparision against it?

Thanks
Huaqi

keith-packard, 2020-12-04:
A short list of key differences might be:

1. Designed for and tested on 32- and 64- bit embedded systems. Newlib (and newlib-nano) are used on both embedded systems and Cygwin, with the bulk of the development focused on improving the Cygwin support. Because the Cygwin version needs to be very careful about ABI stability, many parts of the newlib code base are very difficult to change. Picolibc has no such constraints, so much work has been done to improve the ABI and API for use in embedded systems.

 2. tinystdio. This new stdio implementation borrows some code from avrlibc along with substantial new code. The goal of tinystdio is to remain compatible with relevant ANSI-C and Posix standards while using a very narrow API to the underlying system (it only requires getc and putc) along with doing no internal allocation and using a minimal amount of RAM (16 bytes on a 32 bit system for stdin/stdout/stderr).

 3. Thread-local-storage support. Instead of creating a large data structure containing all possible thread-specific data, Picolibc uses the underlying TLS support from the compiler so that you use only the amount of per-thread storage needed to support the API used by the application. Typically, this means you use only 4 bytes for errno.

 4. Integrated test suite. The old newlib test code, which hasn't worked for years, has been fixed up and is used to test changes to the library. The test suite was used to correct numerous bugs in the math library; those changes have mostly been passed on to newlib.

 5. Bare-metal startup code and linker scripts for RISC-V, ARM and AARCH64 processors. These may not support precisely what your application requires, but these provide a good basis for writing your own startup code and linker scripts. These are used to create bare-metal versions of the test suite which can then be run using QEMU.

 6. Meson-based build system. This is largely used to avoid conflicts with the newlib autotools build system files, but a side effect is that the library builds very quickly.

 7. Narrow and well-defined POSIX OS requirements. Portions of Picolibc which require underlying OS support are documented so that you will  know which functions are necessary based on the API in use. In particular, tinystdio requires *no* POSIX functions to support stdin/stdout/stderr, and instead requires only functions to get and  put single characters to the target device.

 8. Built-in semihosting support for RISC-V, ARM and AARCH64. This allows you to take advantage of existing gdb, openocd and qemu semihosting support for things like debugging output even before you  have a serial port working.
 
 9. Clear BSD licensing. All non-BSD compatible library source code has been removed so that the status of the resulting library is no longer in question.

I did a presentation about picolibc way back in January at LCA: https://lca2020.linux.org.au/schedule/presentation/148/

closed 2020-12-04 by keith-packard

fanghuaqi, 2020-12-04:
Thank you @keith-packard for such detailed information, it looks pretty good to me in embedded software development.

abrodkin, 2023-06-05:
> I did a presentation about picolibc way back in January at LCA: https://lca2020.linux.org.au/schedule/presentation/148/

Unfortunately there's nothing by that link, though looks like its a bit newer version is here: https://keithp.com/picolibc/picolibc-2021-notes.pdf

keith-packard, 2023-06-06:
It appears to only be available on YouTube now? https://www.youtube.com/watch?v=SC6aBezNFFQ

bmcdonnell-fb, 2025-05-20:
Interesting project. I just stumbled upon it today.

@keith-packard you might consider putting the info you shared here somewhere in the documentation. IME newlib-nano is pretty common in the embedded space, and this is the first question I had.

keith-packard, 2025-05-21:
> Interesting project. I just stumbled upon it today.
> 
> [@keith-packard](https://github.com/keith-packard) you might consider putting the info you shared here somewhere in the documentation. IME newlib-nano is pretty common in the embedded space, and this is the first question I had.

Good idea; I've posted PR #1041 with an edited version of this list.