picolibc/picolibc #104: Qemu RISC-V with picolibc semihosting and gdb receives SIGSEGV

aewag, 2020-12-23:
I am setting up Qemu and picolibc for a RISC-V architecture. I am currently looking into the semihosting support for this using Qemu from your feature branch here: https://github.com/keith-packard/qemu/tree/riscv-semihost

I experimented with your hello-world example and was trying to connect with gdb to it to step through the code. For that I firstly added `-S -s` as arguments to Qemu. Using this I continued the execution within gdb and it worked like expected.

I changed the arguments to only providing `-s` as a test and received a segmentation fault. This is something i didn't expect to happen.

I added the backtrace of the SIGSEGV:
```
$ ./run-riscv 
gdb --args /home/alwagner/prog/qemu_riscv/bin/qemu-system-riscv32 -chardev stdio,mux=on,id=stdio0 -semihosting-config enable=on,chardev=stdio0,arg=test-file -mon chardev=stdio0,mode=readline -serial chardev:stdio0 -machine virt,accel=tcg -cpu rv32 -kernel hello-world-riscv.elf -bios none -nographic -s
GNU gdb (Ubuntu 8.1.1-0ubuntu1) 8.1.1
Copyright (C) 2018 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /home/alwagner/prog/qemu_riscv/bin/qemu-system-riscv32...(no debugging symbols found)...done.
(gdb) run
Starting program: /home/alwagner/prog/qemu_riscv/bin/qemu-system-riscv32 -chardev stdio,mux=on,id=stdio0 -semihosting-config enable=on,chardev=stdio0,arg=test-file -mon chardev=stdio0,mode=readline -serial chardev:stdio0 -machine virt,accel=tcg -cpu rv32 -kernel hello-world-riscv.elf -bios none -nographic -s
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
[New Thread 0x7fffec2fe700 (LWP 1376)]
[New Thread 0x7fffebafd700 (LWP 1377)]

Thread 3 "qemu-system-ris" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fffebafd700 (LWP 1377)]
0x0000555555a23d34 in ?? ()
(gdb) bt
#0  0x0000555555a23d34 in  ()
#1  0x0000555555a444a4 in  ()
#2  0x0000555555a8e385 in  ()
#3  0x0000555555a08cfc in  ()
#4  0x00005555559d3908 in  ()
#5  0x0000555555a91818 in  ()
#6  0x0000555555ac3692 in  ()
#7  0x0000555555ac37e7 in  ()
#8  0x0000555555c23506 in  ()
#9  0x00007ffff4f466db in start_thread (arg=0x7fffebafd700) at pthread_create.c:463
#10 0x00007ffff4c6f71f in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:95
(gdb) 
```

And versions for the respective tools used:
```
$ /home/alwagner/prog/gcc_riscv/bin/riscv32-unknown-elf-gcc --version
riscv32-unknown-elf-gcc (GCC) 10.2.0
Copyright (C) 2020 Free Software Foundation, Inc.
This is free software; see the source for copying conditions.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.

$ /home/alwagner/prog/qemu_riscv/bin/qemu-system-riscv32 --version
QEMU emulator version 5.1.50 (v5.1.0-2565-g549e09deae)
Copyright (c) 2003-2020 Fabrice Bellard and the QEMU Project developers

picolibc$ git show
commit 0862d85d3e9ac3f1bd2ca6b7f42e15492fa3ad34 (HEAD -> main, origin/main, origin/HEAD)
Author: Keith Packard <keithp@keithp.com>
Date:   Wed Dec 16 17:56:06 2020 -0800
```

keith-packard, 2020-12-23:
I've reproduced this even on the latest version of the qemu semihosting patches, which are at:

https://github.com/keith-packard/qemu/tree/semihosting

I've proposed a fix in qemu and sent it upstream for review; you can see it here:

https://github.com/keith-packard/qemu/tree/semihost-fix

In any case, using qemu with '-s' but without '-S' probably doesn't do what you want, as it will run without waiting for GDB to connect. But it shouldn't crash :-)

I'll close this bug; we can track progress of the qemu fix by monitoring the qemu-devel mailing list.

closed 2020-12-23 by keith-packard

aewag, 2020-12-24:
Nice, thank you for the quick answer and proposed fix.

>In any case, using qemu with '-s' but without '-S' probably doesn't do what you want, as it will run without waiting for GDB to connect. But it shouldn't crash :-)

Agreed, don't even know why I tried. Just played a bit around. :D

keith-packard, 2020-12-24:
We'll see what the qemu devs think, there are a bunch of places where -s without -S is likely to cause crashes though. Makes me wonder why those options are separate ...

Thanks for the fun bug report, and have a good week.