picolibc/picolibc #105: `error: '_Atomic' does not name a type` with c++ source and gcc (arm-none-eabi)

mlampert, 2021-01-05:
Recent Debian Testing update broke my builds because <stdio.h> includes <stdatomic.h>. This in turn uses the `_Atomic` keyword which is not defined in C++. From what I understand <stdatomic.h> cannot be used in C++ which begs the question what were they thinking.
picolibc-arm-none-eabi 1.4.7-2
gcc-arm-none-eabi 8.3.1

keith-packard, 2021-01-05:
This has been fixed on main; I need to do a release to get that into debian...

Here's the relevant patch: https://github.com/picolibc/picolibc/commit/12d5617f5c8f854c63b9435a8a9f0c306de1d96f

I've been working on better c++ support; I recently got a patch into gcc to make libstdc++ work with picolibc, which should be in gcc version 11 when released. I'll see if I can't get a picolibc release done this week to make it possible to build c++ apps which don't need libstdc++ at least.

closed 2021-01-05 by keith-packard

mlampert, 2021-01-05:
Great - thanks a lot!
And yes, I don't use libstdc++, not a good fit for (my) embedded projects

mlampert, 2021-01-06:
Installed and built from github sources - works like a charm.
Also, never used meson before and yet this was one of the most painless builds ever!
Thanks again!

keith-packard, 2021-01-06:
Awesome! I'm getting ready for another release; should be another day or
so. Thanks for testing, and yes, meson isn't completely terrible, and it
builds so fast with ninja