picolibc/picolibc #107: No atexit handling with picocrt

archi, 2021-01-26:
I just noticed that the `atexit` doesn't seem to properly work, at least not in semihosted mode and with my rv32 configuration (mind that I disabled the `newlib-initfini-array` because CompCert does not support the two necessary attributes):

```
/local/picolibc/meson/meson.py \
    /local/picolibc/picolibc-extras/build.rv32_linux/src \
    /local/picolibc/picolibc-extras/build.rv32_linux/build \
    --prefix=/local/picolibc/compcert_riscv/picolibc \
    --cross-file=/local/picolibc/picolibc-extras/config/rv32_linux/crossfile \
    -Dincludedir=rv32_linux/include \
    -Dlibdir=rv32_linux/lib \
    -Dspecsdir=rv32_linux \
    -Dtests=true \
    -Dmultilib=false \
    -Dpicolib=true \
    -Dpicocrt=true \
    -Dthread-local-storage=false \
    -Dtls-model= \
    -Datomic-ungetc=false \
    -Dnewlib-initfini=false \
    -Dnewlib-initfini-array=false \
    -Dhave-alias-attribute=false \
    -Dwant-math-errno=true
```

Small example:
```
#include <stdlib.h>
#include <stdio.h>

static void callthis() {
  printf("hello\n");
}

int main() {
  atexit(callthis);
  return 0;
}
```

Both the `crt0.S` as well as the `crt0.h` call `_exit`, but the semihost `_exit` does not call `__call_exitprocs`.
The newlib only seems to call `__call_exitprocs` in `cxa_finalize` (`newlib/libc/stdlib/cxa_finalize.c`) and `exit` (`newlib/libc/stdlib/exit.c`). The `exit` calls `_exit`, but unlike the crt0 code it does not check if `_exit` exists.

I see two (or three) options:
1. Call newlib's `exit` instead of `_exit` in `crt0.*` (and move the `if (_exit)` check there as well)
2. Copy the relevant code from `newlib/libc/stdlib/exit.c` to the semihost `exit.c`
3. I'm using/building it wrong ;-)

keith-packard, 2021-01-26:
The whole 'exit' processing thing is difficult to understand for deeply embedded systems -- for most of those, 'exit' has no useful meaning. And, as we're always concerned about memory usage, getting rid of unused code is important. So, for most 'real' uses of picolibc, we don't want any of the exit processing to even be linked into the application.

Which means that crt0 shouldn't force the linking of all of the exit processing code, including the code to handle atexit stuff.

The easy way to make that happen is to just not call exit() from crt0. But, as you've discovered, that breaks lots of stuff.

I'd like to write down what we want and then fix the code. For instance, destructors *are* invoked after main returns, which definitely seems odd now that I look at it.

For embedded applications, main() should never return. In this case, any code after the branch to main is dead and should not be included in the application. Maybe this means that there should be *no* code after the call to main, and that any application wanting to use the semihosting exit functionality should just call exit() directly? That would simplify a bunch of stuff in crt0, and make embedded applications smaller.

keith-packard, 2021-01-26:
oh, compcert doesn't have constructors and destructors? Ok, that's actually helpful -- I was thinking that atexit/on_exit should be processed by a destructor, which would be nice and neat. Without that, perhaps the on_exit/atexit stuff should be called and *then* the destructors.

keith-packard, 2021-01-26:
I've pushed a branch, pico-exit, which removes all exit handling from picocrt so that it's no longer doing part of the job. I've also written a lighter-weight version of atexit/on_exit which removes the dependency on malloc/free and recursive mutexes.

With this version, your application will need to call exit() instead of returning from main. exit() will now invoke all atexit/on_exit handlers and then any destructors (assuming toolchain support).

Let me know if this seems reasonable to you.

archi, 2021-01-27:
I only stumbled over this because we have a test for `atexit()`, and I understand that this is more a theoretical issue since embedded system usually don't return from main. I wouldn't mind calling `exit()` for that one test. And if you said "there is simply no atexit support in picolibc" I'd happily disable that test ;-)

Sadly I don't seem to be able to find your branch? But what you say sounds very reasonable.
I think overall there are just two important things:
1. make sure that constructors/destructors are called properly (for those compilers that support them :laughing:), else people might be in for some bad surprises (I think we can check that box?)
2. document the non-standard behaviour - the atexit is already mentioned in the [build.md](https://github.com/picolibc/picolibc/blob/main/doc/build.md), maybe a sentence or two in the [using.md](https://github.com/picolibc/picolibc/blob/main/doc/using.md) might be a good idea.

archi, 2021-01-27:
Ah, I just saw that you already have the changes in branch main.

99.9% of our tests do a `return 0;` at the end of `main()`, and these now spin forever when running in qemu :( I have to admit I liked the old variant (jump to _start) more ;)

keith-packard, 2021-01-27:
picolibc also has built-in tests for atexit, so anything we do here should get tested.

> Ah, I just saw that you already have the changes in branch main.
> 
> 99.9% of our tests do a `return 0;` at the end of `main()`, and these now spin forever when running in qemu :( I have to admit I liked the old variant (jump to _start) more ;)

heh. At least I added the infinite loop after main returned?

I thought about adding a config option to have crt0 call exit(), but decided that I didn't want to have that kind of variance in the ABI. Could easily revisit that idea and stick that in.

Any reason to *not* just change your tests to call exit() instead?

keith-packard, 2021-01-27:
>     1. make sure that constructors/destructors are called properly (for those compilers that support them laughing), else people might be in for some bad surprises (I think we can check that box?)

Definitely.  Constructors are already part of crt0, and now destructors are part of exit(), but not _exit() (or return from main). Destructors are called *after* atexit/on_exit functions.

>     2. document the non-standard behaviour - the atexit is already mentioned in the [build.md](https://github.com/picolibc/picolibc/blob/main/doc/build.md), maybe a sentence or two in the [using.md](https://github.com/picolibc/picolibc/blob/main/doc/using.md) might be a good idea.

Oh, that is a good idea. I'll go do that.

archi, 2021-01-27:
Mostly that there are many tests (but I believe I can think of some hack to avoid rewriting all of them).

OTOH, since the semihosting defines `_exit`, could we "bring back" the
```
if (_exit)
  _exit(ret)`;
```
part? Remember that `_exit` is defined as

```
extern void
_exit(int) __weak_symbol;
```

in `crt0.h`, and had `.weak _exit` in the `crt0.S`.

That would make my life much easier. Or does that already pull in too much code?

Overall, I'll talk to my colleagues what they think (tomorrow); I can't decide on "our" opinion alone here.

keith-packard, 2021-01-27:
yeah, having it just call _exit() would be simple, and not a huge amount of waste. Still, it's non-standard, and so in some ways not really better than having it infinite loop? The old plan of having it call the destructors and not calling atexit/on_exit stuff seems really wrong though.

I'd rather leave it without a call to _exit so that it is more useful as-is in embedded applications, but let me know what your team thinks tomorrow and we'll go on from there. There are a couple of c++ compatibility fixes queued that I'd like to get into a release for crosstool-ng and zephyr, so I'll probably do a minor release containing these changes in a few days.

archi, 2021-01-28:
Hi Keith, I gave it some more thought and discussed this with my colleagues.
For our internal usage at AbsInt the conclusion is: Worst case we just patch the crt0 locally or create some wrapper for our test cases - but we can life with whatever you decide here.

So, from here on my personal thoughts on the matter :)

The relevant sentences in the C standard (C11, N1570 draft) is 5.1.2.1 2:
> The effect of program termination in a freestanding environment is implementation-
defined.

And 5.1.2.2.3 for hosted:
> If the return type of the main function is a type compatible with int, a return from the
> initial call to the main function is equivalent to calling the exit function with the value
> returned by the main function as its argument;11) reaching the } that terminates the
> main function returns a value of 0. If the return type is not compatible with int, the
> termination status returned to the host environment is unspecified.

I think freestanding is kind of the natural choice for embedded, so it makes sense that picolibc behaves the way it does now. But having both options would do no harm (iff maintainable). And strictly speaking the semihosted variant is a "hosted environment".

I'd therefore propose to add some `#ifdef`s to the crt0 and toggle their behaviour based on a build time define.

1. Per default, the `crt0.o` should build without a call to `exit` for the normal, freestanding execution environment (= status quo as per your current changes).
2. Set some define (e.g. `-D_PICOLIBC_CRT0_SEMIHOST_`) and produce a `crt0-semihosted.o`. That one then calls "exit" on return from main to conform to the hosted environment.

The spec file could then be modified to pick the semihosted variant if `--oslib=semihost` is passed:

```
*startfile:
[...]/lib/%M/crt0%{-oslib=semihost:-semihosted}%O%s
```

_Rant: I'd prefer `crt0%{-oslib=%*}%O%s`, but that adds a space (-> `crt0-semihost .o`) and I couldn't figure out how to prevent that :-(_

I'll prepare a PR for that.
If you like it, you can then merge it, and if you don't like it, I'd be happy to find a variant that makes both of us happy :)
(or I'll just use it internally for our tests if you don't like it at all)

archi, 2021-01-28:
Please disregard the first two commits (a749281 & 5b1d00d). I fixed a few typos, amended my original commit and the force-pushed the changes. Seems GitHub doesn't like that too much.

keith-packard, 2021-01-29:
> _Rant: I'd prefer `crt0%{-oslib=%*}%O%s`, but that adds a space (-> `crt0-semihost .o`) and I couldn't figure out how to prevent that :-(_

I figured it out; just requires putting the %O inside the {} and adding an else clause. Geez the semantics of specs files is hard.

keith-packard, 2021-02-02:
Ok, with #109 and #110 all merged in, I think we can close this issue! Thanks very much for improving the usefulness of the sample crt0 code.

closed 2021-02-02 by keith-packard