picolibc/picolibc #108: Build as meson subproject fails to find picolibc.h

aewag, 2021-01-28:
I want to embed the picolibc lib as a meson subproject. Meson does find picolibc as subprojects, but building fails:

 [1/894] Compiling C object 'subprojects/picolibc/n...ibc/machine/riscv/7ea2ed8@@machine@sta/memset.S.o'
FAILED: subprojects/picolibc/newlib/libc/machine/riscv/7ea2ed8@@machine@sta/memset.S.o 
[...]/riscv32-unknown-elf-clang -Isubprojects/picolibc/newlib/libc/machine/riscv/7ea2ed8@@machine@sta         -Xclang -fcolor-diagnostics -pipe -D_FILE_OFFSET_BITS=64 -Wall -Winvalid-pch -Wextra -Werror -std=c11 -g -ffunction-sections -ftls-model=local-exec --gcc-toolchain=[...]/riscv -march=rv32imc -mabi=ilp32 -mcmodel=medany -include picolibc.h -fno-common --gcc-toolchain=[...]/riscv -march=rv32imc -mabi=ilp32 -mcmodel=medany -fno-builtin -MD -MQ 'subprojects/picolibc/newlib/libc/machine/riscv/7ea2ed8@@machine@sta/memset.S.o' -MF 'subprojects/picolibc/newlib/libc/machine/riscv/7ea2ed8@@machine@sta/memset.S.o.d' -o 'subprojects/picolibc/newlib/libc/machine/riscv/7ea2ed8@@machine@sta/memset.S.o' -c ../subprojects/picolibc/newlib/libc/machine/riscv/memset.S
<built-in>:318:10: fatal error: 'picolibc.h' file not found
#include "picolibc.h"

Building picolibc separately works and the picolibc.h is generated when executing the configure script, so there is probably a misconfiguration on my side or a misunderstanding that I fail to see.

In the top meson.build I integrated picolibc as a subproject:
```
picolibc_proj = subproject('picolibc')
picolibc_dep = picolibc_proj.get_variable('picolibc_dep')
```

Do I have to run the configure step separately?

keith-packard, 2021-01-28:
Ah, I sorted it out -- I added an explicit '-include', 'picolibc.h' to the internal c_args so that every file would be sure to have these definitions available (a few asm files were missing this, in particular). A couple of options to fix this:

 1. Fix the source code to make sure picolibc.h is included everywhere it is needed and remove it from the default c_args.
 2. Split that part of c_args out to a separate var so that wrapping projects don't have that in their compile commmand
 3. Fix the path used to find picolibc.h so that it is relative to the picolibc directory instead of the project root.

keith-packard, 2021-01-28:
I've created a sample application that uses picolibc as a subproject to explore this issue. I think the key is to not use the c_args in your project -- the only thing in the picolibc c_args is this --include option.

Check out https://github.com/picolibc/pico-hello and see if that helps you get this working.

closed 2021-01-28 by keith-packard

aewag, 2021-02-01:
Thank you! That example code really helped and was a good starter for better understanding.