picolibc/picolibc #115: Making picolibc's spec file less dependent on it's location on the filesystem

archi, 2021-03-26:
Since I don't always control the location of picolibc in our internal CI, I had to somehow make the spec file "less" position dependent.

tl;dr: My variant is okay for us, but not exactly nice for the general audience. Maybe you have an idea how to make picolibc truly "position independent"?

As the first step, my (perl) compile script "fixes" some paths to known values:
```perl
# just a simplified excerpt; collect our output dirs:
my $outdir_base = "/tmp/foo/bar";
my $config = "rv32_linux";

my $outdir = catfile($outdir_base, $config);

system("meson",
    # [...]
    "--prefix=$outdir_base",
    "-Dincludedir=" . catfile($config, "include"),
    "-Dlibdir=" . catfile($config, "lib"),
    "-Dspecsdir=$config",
    # [...]
```
So the spec file ends up with several instances of "/tmp/foo/bar/rv32_linux".

The compile script then post-processes the spec file to add an _optional_ "--picolibc-base" parameter:
```perl
foreach my $line (@spec_lines) {
    $line =~ s@$outdir([^ ]*)@%{-picolibc-base=*:%*$1; :$outdir$1}@g;
    print $out $line;
}
```
I then configure CompCert's driver to pass `--picolibc-base=%<compcert_ini_dir>%/../picolibc/rv32_linux` to the preprocessor and linker, so the user just calls `ccomp -t rv32-linux-picolibc`. Not passing the `--picolibc-base` (like the tests do) causes falling back to the original `$outdir`.

This isn't a good solution (I only did it because for us the ugliness is hidden in the generated `compcert.ini`), but maybe others would also profit from "something like this". **Maybe you have an idea how to make picolibc truly "position independent"?**
Maybe it's possible to extract the picolibc base path from the `--specs` parameter, but that would require to always hardcode some paths (or add logic to derive the relative locations in the `meson.build`).

Of course this exact approach could also be ported to the `meson.build`, but something less explicit would probably be better ;)

I am not sure how continue here or what do propose in a commit, so let me know what you think or would prefer :)

keith-packard, 2021-03-26:
Picolibc does have the sysroot-install parameter which can be paired with a gcc version that has a default sysroot set so that picolibc is installed relative to the compiler; if the compiler moves, then the picolibc files would move along with it. Another option could be to allow the user to override the location with a command line parameter passed to the compiler. I looked at using environment variables, but the specs file semantics don't allow for optional environment variables.

If not for the multilib support, you could simply override the location using -isystem and -L command line parameters.

keith-packard, 2021-03-26:
Is #116 useful for you?

archi, 2021-03-29:
I've adjusted my build-scripts to #116 and it works fine :)

archi, 2021-03-31:
Thanks for merging the changes. Imho we can close this issue then.

closed 2021-03-31 by archi

keith-packard, 2021-03-31:
thanks for closing; I forgot to do this when I merged...