picolibc/picolibc #119: [Feature request] Ability to disable Complex support in libm

ReimuNotMoe, 2021-05-23:
There are some compiler for MCUs that doesn't support the `_Complex` type, such as Microchip's XC16.

Could you add an option to disable building the `newlib/libm/complex` directory?

Thanks.

keith-packard, 2021-05-24:
This should already happen automatically -- picolibc configuration checks if this code compiles and only enables the complex APIs if it does:

        float _Complex test(float _Complex z) { return z; }

Can you check and see if `HAVE_COMPLEX` is defined in the generated `picolibc.h` file?

keith-packard, 2021-06-08:
I'm going to close this; I *think* everything should  be correct. Please re-open with a comment if you can construct a sample which doesn't work.