picolibc/picolibc #127: error: expected declaration specifiers or ‘...’ before numeric constant in expansion of macro ‘__ssp_bos0’

dbungert, 2021-06-03:
Greetings!

I was investigating build failures of picolibc on Ubuntu Impish, and found
that I was able to produce similar build failures using the raw picolibc source
as well.

Build system: Ubuntu amd64 container in lxd (tested focal, hirsute, impish)

Build steps - main branch:
  sudo apt install build-essential ninja-build meson git
  git clone https://github.com/picolibc/picolibc
  (current commit: f909f310f7c0fcd22234eacef15a739ceaf09277)
  cd picolibc
  mkdir amd64
  cd amd64
  ../scripts/do-amd64-configure
  ninja

Failure observed:
```
In file included from ../newlib/libc/include/ssp/strings.h:34,
                 from ../newlib/libc/include/strings.h:86,
                 from ../newlib/libc/include/string.h:56,
                 from ../newlib/libc/string/bcopy.c:39:
../newlib/libc/include/ssp/ssp.h:47:52: error: expected declaration specifiers or ‘...’ before numeric constant
   47 | #define __ssp_bos0(ptr) __builtin_object_size(ptr, 0)
      |                                                    ^
../newlib/libc/include/ssp/strings.h:40:7: note: in expansion of macro ‘__ssp_bos0’
   40 |     ((__ssp_bos0(dst) != (size_t)-1) ? \
      |       ^~~~~~~~~~
../newlib/libc/string/bcopy.c:43:1: note: in expansion of macro ‘bcopy’
   43 | bcopy (const void *b1,
      | ^~~~~
In file included from ../newlib/libc/include/strings.h:86,
                 from ../newlib/libc/include/string.h:56,
                 from ../newlib/libc/string/bcopy.c:39:
../newlib/libc/include/ssp/strings.h:40:23: error: expected ‘)’ before ‘!=’ token
   40 |     ((__ssp_bos0(dst) != (size_t)-1) ? \
      |                       ^~
../newlib/libc/string/bcopy.c:43:1: note: in expansion of macro ‘bcopy’
   43 | bcopy (const void *b1,
      | ^~~~~
../newlib/libc/include/ssp/strings.h:40:38: error: expected ‘)’ before ‘?’ token
   40 |     ((__ssp_bos0(dst) != (size_t)-1) ? \
      |                                      ^
../newlib/libc/string/bcopy.c:43:1: note: in expansion of macro ‘bcopy’
   43 | bcopy (const void *b1,
      | ^~~~~
```
Sample ubuntu build log (not using the above build steps):
https://launchpad.net/ubuntu/+source/picolibc/1.5.1-2/+build/21470303/+files/buildlog_ubuntu-impish-amd64.picolibc_1.5.1-2_BUILDING.txt.gz

keith-packard, 2021-06-05:
Oh, this looks like it's building picolibc with SSP_FORTIFY_LEVEL set. You'll need to adjust the build parameters to disable this somehow as Ubuntu enables stack smashing protection by default (which is a good thing, but must be disabled when building the C library which includes the code to implement stack smashing protections...)

keith-packard, 2021-06-05:
I got some help from Kees Cook this morning and added some compiler options to disable the stack protector and fortify source bits when building picolibc.  Fixed on main now.

closed 2021-06-05 by keith-packard

dbungert, 2021-06-07:
Keith and Kees, thanks.
I can confirm that this is building now, both in my test case, and in a modified package.