picolibc/picolibc #129: ARM semihosting not working correctly

rdiez, 2021-06-06:
I have written a test case that calls abort(), but QEMU was still exiting with status code 0.
After investigating the issue, I have arrived at this routine:

```
sys_semihost_exit_extended(uintptr_t code)
{
  struct {
          sh_param_t      field1;
          sh_param_t      field2;
  } arg = {
          .field1 = ADP_Stopped_ApplicationExit,
          .field2 = code
  };
  (void) sys_semihost(SYS_EXIT_EXTENDED, (uintptr_t) &arg);

```
I am emulating an ARM Cortex-M3 CPU, so _sys_semihost_ is a very short assembly routine that runs "bkpt    0x00ab".  The ARM semihosting interface ANGEL specifies that the operation and its arguments must be passed in registers R0, R1 and R2, but that is not the case in the code above.

The first function parameter (SYS_EXIT_EXTENDED) does land in R0, which is why it mostly works, but R1 contains a memory address (a pointer) and R2 will probably contain rubbish. That's the reason why the exit code does not make it.

keith-packard, 2021-06-08:
which semihosting spec are you referencing? This code is designed to support semihosting version 2.0:

https://developer.arm.com/documentation/100863/0300/Semihosting-operations/SYS-EXIT-EXTENDED--0x20-?lang=en

In this version, the parameters are passed in this parameter block in memory, which is referenced by r1.

Perhaps you're using something that supports an older version of the spec? Both qemu and openocd support this newer version. And there are test cases using qemu which this code does work with?

rdiez, 2021-06-09:
I probably got confused somehow. The documentation for SYS_EXIT_EXTENDED does work with a pointer. I hope that ARM did not break compatibility between semihosting versions.

I am using QEMU version 4.2.1 (Debian 1:4.2-3ubuntu6.16) that comes with Ubuntu 20.04.2. This version should support SYS_EXIT_EXTENDED, but it seems to exit with a 0 status code when the test application calls abort().

I looked at Picolibc's scripts/run-arm , but it does not look like it would be discarding QEMU's exit code.

keith-packard, 2021-06-09:
Looks like the abort() path tried to exit with status sig << 8 and that got truncated to 8 bits somewhere along the line. I've changed the exit code in this case to 128 + sig, which should avoid that truncation. I've pushed this fix to the abort-failure  branch, if that passes tests, I'll merge to main. Thanks for the bug report!

keith-packard, 2021-06-09:
All merged!