picolibc/picolibc #130: [Compilation error] using picolibc with math.h on stm32

matthijs040, 2021-06-07:
Hi,

I am in the process of switching between newlib-nano and picolibc for a project on an stm32f303 microcontroller.
I am including math.h for using the tan2(float, float) function. This was run and tested using newlib-nano initially and errors when building the same code under picolibc.

The compiler error refers to this function:
```c
/*
 * Single precision functions
 */

__declare_arm_macro(float)
__ieee754_sqrtf(float x)
{
	float result;
#if __ARM_ARCH >= 6
	asm ("vsqrt.f32 %0, %1" : "=w" (result) : "w" (x));
#else
	/* VFP9 Erratum 760019, see GCC sources "gcc/config/arm/vfp.md" */
	asm ("vsqrt.f32 %0, %1" : "=&w" (result) : "w" (x));
#endif
	return result;
}
```
And states:
```bash
In file included from /usr/local/picolibc/arm-none-eabi/include/math.h:662,
                 from inc/acc_att_est.h:31,
                 from main.c:5:
/usr/local/picolibc/arm-none-eabi/include/machine/math.h: In function '__ieee754_sqrtf':
/usr/local/picolibc/arm-none-eabi/include/machine/math.h:142:25: error: expected ')' before ':' token
  142 |  asm ("vsqrt.f32 %0, %1" : "=w" (result) : "w" (x));
      |                         ^~
      |                         )
make: *** [rules.mk:214: *.o] Error 1
```
I am unsure what this error means as I have never used inline assembly in a C project.
Is there an issue in this function or am i compiling/using it in a wrong way that causes it?

rdiez, 2021-06-07:
I guess you mean atan2() or atan2f().

I am building a test program for a simulated ARM Cortex-M3, and the compiler does not complain about those functions. What compilation flags are you using to build your firmware and/or picolibc?

matthijs040, 2021-06-07:
I meant atan2 yeah. woops :)
My makefile is a modified libopencm example so it's a little big...
<details>   <summary> Makefile: </summary>

```Makefile
##
## This file is part of the libopencm3 project.
##
## Copyright (C) 2009 Uwe Hermann <uwe@hermann-uwe.de>
## Copyright (C) 2010 Piotr Esden-Tempski <piotr@esden.net>
## Copyright (C) 2013 Frantisek Burian <BuFran@seznam.cz>
##
## This library is free software: you can redistribute it and/or modify
## it under the terms of the GNU Lesser General Public License as published by
## the Free Software Foundation, either version 3 of the License, or
## (at your option) any later version.
##
## This library is distributed in the hope that it will be useful,
## but WITHOUT ANY WARRANTY; without even the implied warranty of
## MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
## GNU Lesser General Public License for more details.
##
## You should have received a copy of the GNU Lesser General Public License
## along with this library.  If not, see <http://www.gnu.org/licenses/>.
##

# Be silent per default, but 'make V=1' will show all compiler calls.
ifneq ($(V),1)
Q		:= @
NULL	:= 2>/dev/null
endif

###############################################################################
# Executables

PREFIX		?= arm-none-eabi-

CC		:= $(PREFIX)gcc --specs=picolibc.specs
CXX		:= $(PREFIX)g++
LD		:= $(PREFIX)gcc --specs=picolibc.specs
AR		:= $(PREFIX)ar
AS		:= $(PREFIX)as
OBJCOPY		:= $(PREFIX)objcopy
OBJDUMP		:= $(PREFIX)objdump
GDB		:= $(PREFIX)gdb
STFLASH		= $(shell which st-flash)
STYLECHECK	:= /checkpatch.pl
STYLECHECKFLAGS	:= --no-tree -f --terse --mailback
STYLECHECKFILES	:= $(shell find . -name '*.[ch]')
OPT		:= -Og
DEBUG		:= -ggdb3
CSTD		?= -std=c18

###############################################################################
# Source files

define uniq =
  $(eval seen :=)
  $(foreach _,$1,$(if $(filter $_,${seen}),,$(eval seen += $_)))
  ${seen}
endef

OBJS		+= ./*.o

ifeq ($(strip $(OPENCM3_DIR)),)
# user has not specified the library path, so we try to detect it

# where we search for the library
LIBPATHS := ./libopencm3 ../../../../libopencm3 ../../../../../libopencm3 ../../lib/libopencm3

[... 195 lines cut ...]
		$(NULL)
endif
else
%.flash: %.elf
	@printf "  GDB   $(*).elf (flash)\n"
	$(GDB) --batch \
		   -ex 'target extended-remote $(BMP_PORT)' \
		   -x $(EXAMPLES_SCRIPT_DIR)/black_magic_probe_flash.scr \
		   $(*).elf
endif

.PHONY: images clean stylecheck styleclean elf bin hex srec list

-include $(OBJS:.o=.d)

```
</details>
<details>   <summary> The output of the make command is: </summary>

```bash
arm-none-eabi-gcc --specs=picolibc.specs -Og -std=c18 -ggdb3 -mcpu=cortex-m4 -mthumb -mfloat-abi=hard -mfpu=fpv4-sp-d16 -Wextra -Wshadow -Wimplicit-function-declaration -Wredundant-decls -Werror -fno-common -ffunction-sections -fdata-sections  -MD -Wall -Wundef -Wno-unused-function -DSTM32F3 -DSTM32F3 -DSTM32F3CCM -DSTM32F303RE -I../../lib/libopencm3//include -c *.c ./src/*.c 
```
</details>

My installation of picolibc has been a while ago. I think i left most things default but I'm not sure how to check that.

rdiez, 2021-06-07:
There is nothing obviously wrong in the information you supplied. I'm afraid I don't know enough about Picolibc yet to help you further.

archi, 2021-06-08:
The problem is your `-std=c18`, as set via the
> CSTD		?= -std=c18
in your Makefile.

My compiler is a bit older, so I used `-std=c11`, but this show the same issue.
However, there should be a warning like this

```
x.c: In function 'test':
x.c:8:5: warning: implicit declaration of function 'asm' [-Wimplicit-function-declaration]
     asm ("vsqrt.f32 %0, %1" : "=w" (result) : "w" (x));
```

[A quick duckduckgo search finds this on stack overflow](https://stackoverflow.com/questions/18574943/gcc-inline-assembler-implicit-function-declaration-on-powerpc-arch): Picolibc is using GNU's extended inline assembly syntax. So you might want to use `-std=gnu18` (I am not 100% sure if that works, but `-std=gnu11` did the trick for me).

matthijs040, 2021-06-08:
That was it! I was worried it was something about the C standard but did not try every option it seems!
Thanks so much for your help. My code now does the same it did under newlib while being ~8kB smaller.
Considering I want to use it on a board with only 32kB's of flash that opens up a lot of room for cool stuff.

keith-packard, 2021-06-08:
I've pushed a change to use __asm__ instead of asm or __asm everywhere; this should let the header files be compiled under --std=c18 now. Picolibc cannot build in that mode at this point; there are still a few more changes necessary. I'll close this (as this should fix the issue). Please re-open if you find something that doesn't work yet.

closed 2021-06-08 by keith-packard