picolibc/picolibc #138: mallinfo uordblks not 0 after freeing everything

rdiez, 2021-06-15:
I am trying to automatically detect memory leaks in my firmware. I am calling mallinfo() on termination and checking whether field uordblks is 0.

It is not actually documented that uordblks should go back to 0 when the user has freed all memory, but I guess that is the usual expectation. See for example the comment here:

https://stackoverflow.com/questions/50589448/memory-leaks-detection-mallinfo-heapwalk

That seems to be the case with Newlib's nano-mallocr.c implementation.

With Picolibc, I wrote a minimal firmware that does the following:

  free( malloc( 123 ) );

But then the final uordblks was 4.

I added the following to test program test/malloc_stress.c :

```
<insert at the beginning of main()>
const struct mallinfo mi_start = mallinfo();
printf( "Begin uordblks: %zu\n", mi_start.uordblks );

```

```
<insert at the end of main()>
const struct mallinfo mi_end = mallinfo();
printf( "End uordblks: %zu\n", mi_end.uordblks );

```
And the final uordblks was 4 too.

This could be the extra padding in newlib/libc/stdlib/nano-mallocr.c next to this comment:

`/* p is not aligned, ask for a few more bytes so that we have`

keith-packard, 2021-06-21:
I've stuck a fix and a test on the fix-uordblks branch. Feel free to review PR https://github.com/picolibc/picolibc/pull/139 if this fixes the bug for you.

rdiez, 2021-06-21:
I am not doing much malloc/free at the moment with Picolibc, but I applied your changes and the limited test case I have now works fine.

keith-packard, 2021-06-21:
Glad to know this is fixed for you. I've merged that PR to main. Also, yeah, malloc is not something one typically exercises in embedded systems. Best to have fewer failure modes.

closed 2021-06-21 by keith-packard