picolibc/picolibc #147: How to create an ABI compatible __iob in application code

kammce, 2021-07-05:
Is there a way to create the __iob[] array in application code rather than creating my own custom oslib? I tried cracking away at it yesterday to no avail. I keep getting that the __file object is incomplete.

keith-packard, 2021-07-06:
Yup, you should be able to create one; I've got several applications that do, along with a bunch of RTOS ports.

Here's an example:

```
#include <stdio.h>
static FILE __stdio = FDEV_SETUP_STREAM(my_own_putc, my_own_getc, my_own_flush, _FDEV_SETUP_RW);

FILE *const __iob[3] = { &__stdio, &__stdio, &__stdio };
```

This is documented in [doc/os.md](https://github.com/picolibc/picolibc/blob/main/doc/os.md), please let us know if that isn't clear or is missing some details.

kammce, 2021-07-06:
It seems that arm-none-eabi-g++ cannot find the `FDEV_SETUP_STREAM` and `_FDEV_SETUP_RW` macros when I run it. I get this:

```
simple.cpp:30:28: error: '__stdio' was not declared in this scope; did you mean 'stdin'?
   30 | FILE * const __iob[3] = { &__stdio, &__stdio, &__stdio };
      |                            ^~~~~~~
      |                            stdin
simple.cpp:30:38: error: '__stdio' was not declared in this scope; did you mean 'stdin'?
   30 | FILE * const __iob[3] = { &__stdio, &__stdio, &__stdio };
      |                                      ^~~~~~~
      |                                      stdin
simple.cpp:30:48: error: '__stdio' was not declared in this scope; did you mean 'stdin'?
   30 | FILE * const __iob[3] = { &__stdio, &__stdio, &__stdio };
      |                                                ^~~~~~~
      |                                                stdin
simple.cpp:33:63: error: '_FDEV_SETUP_RW' was not declared in this scope
   33 |     FDEV_SETUP_STREAM(sample_putc, sample_getc, sample_flush, _FDEV_SETUP_RW);
      |                                                               ^~~~~~~~~~~~~~
simple.cpp:33:5: error: 'FDEV_SETUP_STREAM' was not declared in this scope
   33 |     FDEV_SETUP_STREAM(sample_putc, sample_getc, sample_flush, _FDEV_SETUP_RW);
      |     ^~~~~~~~~~~~~~~~~
simple.cpp:32:13: warning: '__stdio' defined but not used [-Wunused-variable]
   32 | static FILE __stdio =
      |             ^~~~~~~
```

kammce, 2021-07-06:
This is the command I'm using (it has a lot of flag, sorry about that)

```
arm-none-eabi-g++ -fexceptions --exceptions -ffunction-sections -fdata-sections -fdiagnostics-color -fno-rtti -fno-threadsafe-statics -fno-omit-frame-pointer -ffreestanding -Wno-main -Wall -Wformat=2 -Wno-uninitialized -Wnull-dereference -Wold-style-cast -Woverloaded-virtual -Wsuggest-override -Wno-psabi -lgcc -Wl,--gc-sections -Wl,--print-memory-usage -Wl,--print-memory-usage --specs=picolibcpp.specs --oslib=semihost --crt0=hosted -T ./library/liblpc40xx/platform/default.ld -g -std=c++20 -I ./library -Og -D PLATFORM=lpc40xx -mcpu=cortex-m4 -mthumb -mfloat-abi=hard -mfpu=fpv4-sp-d16 -mtpcs-frame -mtpcs-leaf-frame -D__FPU_PRESENT -Xlinker --defsym=__heap_start=0x20000000 -Xlinker --defsym=__heap_end=0x20008000 -Xlinker --defsym=__heap_size=0x8000 main.cpp -o ./build/lpc40xx/./lpc40xx.main.cpp.elf
```

kammce, 2021-07-06:
The code looks like:

```C++
#include <stdio.h>
static int sample_putc(char c, FILE * file)
{
  (void)file;
  return c;
}

static int sample_getc(FILE * file)
{
  (void)file; /* Not used in this function */
  return 0;
}

static int sample_flush(FILE * file)
{
  /* This function doesn't need to do anything */
  (void)file; /* Not used in this function */
  return 0;
}

FILE * const __iob[3] = { &__stdio, &__stdio, &__stdio };

static FILE __stdio =
    FDEV_SETUP_STREAM(sample_putc, sample_getc, sample_flush, _FDEV_SETUP_RW);

int main() {
  // ...
}
```

kammce, 2021-07-06:
Ok so I used an absolute path to the stdio.h file from the application and go this:

```
In file included from gcc-arm-none-eabi-picolibc/arm-none-eabi/include/stdlib.h:18,
from gcc-arm-none-eabi-picolibc/arm-none-eabi/include/c++/10.2.1/cstdlib:75,
from gcc-arm-none-eabi-picolibc/arm-none-eabi/include/c++/10.2.1/bits/stl_algo.h:59,
from gcc-arm-none-eabi-picolibc/arm-none-eabi/include/c++/10.2.1/vector:62,
from gcc-arm-none-eabi-picolibc/arm-none-eabi/include/c++/10.2.1/functional:62,
from libcore/platform/syscall.hpp:3,
from simple.cpp:3:
gcc-arm-none-eabi-picolibc/arm-none-eabi/include/sys/reent.h:292:26: error: conflicting declaration 'typedef struct __sFILE __FILE'
  292 | typedef struct __sFILE   __FILE;
      |                          ^~~~~~
```

I removed all of the path information from the output but everything else is basically the same. What I'm seeing here is that, because all of my code uses the C++ style C headers `<cstdio>` headers vs the `stdio.h` style headers, its pulling in the <cstdio> code which is causing a conflict. I don't see any of the C++ style includes in picolibc so I'm guessing that G++ is failing to find them in the picolibc directory and falling back to the newlib ones.

keith-packard, 2021-07-06:
Ah, you're using c++. There's a problem with the gcc standard C++ library where it uses `#include_next` to force a specific ordering of the header files. If you have newlib installed on your system, then that will find the newlib headers instead of the picolibc headers.

I've got a work-around for this issue that I included in the crosstool-ng build environment which replaces the `#include_next` with regular `#include` statements so that the picolibc headers are used instead of the newlib headers. The alternative is to remove newlib from your system.

Here's the patch that crosstool-ng includes:
https://github.com/crosstool-ng/crosstool-ng/commit/28101b82564f504710919bed39a064356c3f13d6

You might consider using crosstool-ng to create a toolchain as that works well with c++ and picolibc.

kammce, 2021-07-06:
Ah ok. Let me try and remove newlib and first and see if that resolves this issue. I don't see a reason to keep newlib or newlib nano on my distribution of arm gcc if I have picolibc.

kammce, 2021-07-06:
So in my circumstance, I want to be able to use the offical arm gcc for linux x86, linux arm, mac, and windows. I can apply picolibc on top of this and utilize all of the benefits of using it. Except for the __iob[] feature meaning that I cannot implement my own STD I/O.  So creating my own toolchain for each is undesirable. Is there some combination of GCC flags that I can do to ensure that the picolibc headers are used and that everything resolves?

I tried this `-nostdinc` and it got rid of all of the std includes I didn't want, but now I get this error:

```
gcc-arm-none-eabi-picolibc/arm-none-eabi/picolibc/arm-none-eabi/include/sys/config.h:34:10: fatal error: float.h: No such file or directory
   34 | #include <float.h>
```

My list of includes looks correct though (meaning just including my library directory and picolibc's include directory):
```
GGC heuristics: --param ggc-min-expand=100 --param ggc-min-heapsize=131072
#include "..." search starts here:
#include <...> search starts here:
 starter/library
 packages/gcc-arm-none-eabi-picolibc/bin/../arm-none-eabi/picolibc/arm-none-eabi/include
End of search list.
```

keith-packard, 2021-07-06:
yeah, you can't use -nostdinc as that removes the GCC headers from the search path too. You could probably figure out where that directory is and add that in yourself. Did you have any luck after removing newlib and newlib-nano?

kammce, 2021-07-06:
I'm not super familiar with the internals of the toolchain. I started hacking away at parts and all it did was make a mess. I definitely doing something wrong. Would you know what sections I need to pull out in order to "remove" newlib and newlib-nano? 

And thank you so much for taking the time to help me out with this.

kammce, 2021-07-06:
I also tried this:

```
extern "C"
{
  struct psuedo_file
  {
    uint16_t unget; /* ungetc() buffer */
    uint8_t flags;  /* flags, see below */
    int (*put)(char,
               struct psuedo_file *); /* function to write one char to device */
    int (*get)(
        struct psuedo_file *); /* function to read one char from device */
    int (*flush)(struct psuedo_file *); /* function to flush output to device */
  };

  int sample_putc(char c, psuedo_file * file)
  {
    (void)file;
    return c;
  }

  int sample_getc(psuedo_file * file)
  {
    (void)file; /* Not used in this function */
    return 0;
  }

  int sample_flush(psuedo_file * file)
  {
    /* This function doesn't need to do anything */
    (void)file; /* Not used in this function */
    return 0;
  }

  struct psuedo_file __stdio = {
    .flags = 0x0001 | 0x0002,
    .put   = sample_putc,
    .get   = sample_getc,
    .flush = sample_flush,
  };

  [[gnu::used]] psuedo_file * const __iob[3] = { &__stdio, &__stdio, &__stdio };
}
```

But got:

```
packages/gcc-arm-none-eabi-picolibc/bin/../lib/gcc/arm-none-eabi/10.2.1/../../../../arm-none-eabi/bin/ld: packages/gcc-arm-none-eabi-picolibc/bin/../arm-none-eabi/picolibc/arm-none-eabi/lib/thumb/v7e-m+fp/hard/libc.a(printf.c.o): in function `printf':
/home/keithp/src/picolibc/builds/build-arm/../../newlib/libc/tinystdio/printf.c:46: undefined reference to `__iob'
```

I even call the `__iob[0]->put('c', nullptr)` to make sure it gets added to the final binary.

keith-packard, 2021-07-06:
In debian and ubuntu, newlib and newlib-nano are separate packages and can be removed without removing the compiler itself.

```$ sudo apt remove libnewlib-arm-none-eabi libnewlib-nano-arm-none-eabi```

However, if you're using c++, I fear you're also using libstdc++, which depends on libc. There are source patches upstream in gcc which makes it work with picolibc, but that requires that it be built against picolibc. I'd encourage you to reconsider using crosstool-ng to build a toolchain for your project instead of attempting to use the distribution packaged version; it's reasonably easy to build, and doesn't require superuser privileges.

kammce, 2021-07-06:
The `apt remove` method doesn't work for my use case since I didn't install via `apt-get`. 

So question about the crosstool-ng, does their "candian" build allow me to use my Linux machine to build a toolchain for OSX that build files for ARM?

keith-packard, 2021-07-07:
Oh, you want to build on OSX. It looks like crosstool-ng has given up trying to support that platform, and I doubt it could support building a toolchain on Linux that ran on Mac OS X.

I wonder if you added the C++ headers *before* the picolibc headers if that would help. The whole libstdc++ install and configure is such a mess; having it be part of the compiler but depending on libc introduces circular dependencies which require building the compiler multiple times and then the include paths are just scrambled.

You should be able to define your own __iob struct by placing it in a regular C file, instead of C++, which should avoid using any of the C++ headers.

kammce, 2021-07-07:
Oh! Putting the contents into a regular C file did it for me. I'll close this out, but I think I'll still look in to ways I can make this work on C++ just because it makes my code a little easier to work with. Otherwise, I can implement what newlib does with their `_write`, `_read` functions that their internal libraries and have my __iob call those. 

I'll report back any findings I have or report back that I couldn't figure it out. 😅

Cheers! And thank you for your help!

closed 2021-07-07 by kammce

keith-packard, 2021-07-07:
It all works fine for newlib because newlib is the 'system' libc, hence placed correctly in the #include references to resolve the libstdc++ #include_next links.

Glad you got something working!

kammce, 2021-07-07:
Actually, now that I think about it, how much of a problem is this? Is this simply swapping out header files or do some other binaries get swapped out as well? Is my system in some blend between newlib and picolibc? If its mostly localized to just the stdio stuff then I think I can work around this but if its more, then I should probably invest some time to ripping out newlib and newlib-nano to be sure. What do you think?

keith-packard, 2021-07-07:
Any time you're using C++ headers that use '#include_next', you'll end up getting the newlib headers instead of the picolibc ones. Mostly that should be OK as the definitions are similar.

keith-packard, 2021-07-07:
one way to avoid the issue is to go edit the libstdc++ headers on your system, replacing instances of '#include_next' with '#include'

kammce, 2021-07-07:
> one way to avoid the issue is to go edit the libstdc++ headers on your system, replacing instances of '#include_next' with '#include'

Let me try this and get back.

kammce, 2021-07-08:
I couldn't get it to work by replacing the #include_next. Seems like it was still finding the newlib headers. 

What did work this morning was replacing all of the files in the `gcc-arm-none-eabi/arm-none-eabi/include` with the contents of the `gcc-arm-none-eabi/arm-none-eabi/picolibc/arm-none-eabi/include` and not touching the C++ includes. And now I can create an __iob structure using the defines. Although I get a "flag initialized out of order" error when I use `FDEV_SETUP_STREAM` so I used `fdev_setup_stream` instead for testing. C++ has a sensitivity to the order of initialization for designated initializers so it gives an error. But I was able to fix this by making this change in `stdio.h`: 

```
#define FDEV_SETUP_STREAM(p, g, fl, f)		\
	{ \
		.flags = f, \
		.put = p, \
		.get = g, \
		.flush = fl, \
	}
```

kammce, 2021-07-08:
I feel like there must be some way to picolibc to the start of the includes search list, which would fix this issue. Or somehow in the specs file, remove the default newlib C libary from the includes list. I think either one of those should work.

But at least for my needs, simply taking the prebuilt binaries for all of the systems I need to support and replacing the includes with the picolibc is a pretty easy, non-laborious, quick task that I can also probably automate. So I may stick with this as I ramp up my project.

Thanks again for your help and for this project!

keith-packard, 2021-07-08:
> I couldn't get it to work by replacing the #include_next. Seems like it was still finding the newlib headers.
>
> What did work this morning was replacing all of the files in the [...]

If you want to provide a patch for the stdio.h issue, please submit a PR with the change; I'd appreciate that as you can test to make sure it works correctly.

keith-packard, 2021-07-08:
> I feel like there must be some way to picolibc to the start of the includes search list, which would fix this issue. Or somehow in the specs file, remove the default newlib C libary from the includes list. I think either one of those should work.

There's no way to remove include elements in the .specs file as far as I've found. As to the ordering issue, the picolibc headers need to come after the libstdc++ headers but before the newlib headers, and there's no option available to do that. Feel free to experiment with editing the .specs file in place to see what you can get working, but it's a bit of a mess as the newlib headers are privileged as the 'system' libc.

The only way to avoid conflict between newlib and picolibc would be to remove newlib from the environment and make picolibc the system libc. And that means convincing your toolchain supplier to do that...

> But at least for my needs, simply taking the prebuilt binaries for all of the systems I need to support and replacing the includes with the picolibc is a pretty easy, non-laborious, quick task that I can also probably automate. So I may stick with this as I ramp up my project.

Awesome! Glad you got this working.

kammce, 2021-07-08:
> If you want to provide a patch for the stdio.h issue, please submit a PR with the change; I'd appreciate that as you can test to make sure it works correctly.

When I get a free moment, I'll make a PR 😄 Probably this weekend. Cheers!

keith-packard, 2021-07-08:
Awesome. I'll wait until Monday to do a 1.7.1 release then as I'd like to have this change in place, along with another C++ requirement (__cxa_atexit).