picolibc/picolibc #158: How to disable float support in printf

michalsieron, 2021-08-12:
How to properly disable float support in `printf` when using tinystdio?

I read [doc/printf.md](https://github.com/picolibc/picolibc/blob/main/doc/printf.md), but it didn't help.

I tried adding `'--specs=picolibc.specs', '-DPICOLIBC_INTEGER_PRINTF_SCANF'` to `c_args` in cross file, but then meson fails checking compiler version, because it can't find picolibc.specs file, as it isn't created yet.

Then I added `'-Wl,--defsym=vfprintf=__i_vfprintf', '-Wl,--defsym=vfscanf=__i_vfscanf'` to `c_link_args`, but it also didn't help.

Finally I found, that [tinystdio uses `PRINTF_LEVEL`](https://github.com/picolibc/picolibc/blob/main/newlib/libc/tinystdio/stdio_private.h#L36) to decide whether to support floats or not.

So right now, I added  `'-DPRINTF_LEVEL=1'` to `c_args` in cross file, but then it warns about redefining it
```
[580/932] Compiling C object newlib/libc/tinystdio/libtinystdio.a.p/vfiprintf.c.o
../newlib/libc/tinystdio/vfiprintf.c:33: warning: "PRINTF_LEVEL" redefined
   33 | #define PRINTF_LEVEL PRINTF_STD
      |
<command-line>: note: this is the location of the previous definition
[602/932] Compiling C object newlib/libc/tinystdio/libtinystdio.a.p/vfprintff.c.o
../newlib/libc/tinystdio/vfprintff.c:33: warning: "PRINTF_LEVEL" redefined
   33 | #define PRINTF_LEVEL PRINTF_FLT
      |
<command-line>: note: this is the location of the previous definition
```

keith-packard, 2021-08-19:
Oh, you don't disable it while building picolibc, but instead while building your application. Stick the -DPICOLIBC_INTEGER_PRINTF_SCANF in your project CFLAGS along with --specs=picolibc.specs and it will map printf to the integer-only version while linking. Picolibc gets built with three different versions of the printf code so you don't have to decide until you use it.

michalsieron, 2021-08-19:
Can I get it to work with LTO enabled for picolibc? When I try doing that, it seems that it has the same size as if floats are enabled.
Space gained by enabling LTO for picolibc (when I was using my previous method) isn't anything big (only 0.05 KB), but I am just curious.

keith-packard, 2021-08-19:
I've never tried LTO, so I'm not sure. You may need to explicitly specify the symbol remapping which picolibc.specs does when you use -DPICOLIBC_INTEGER_PRINTF_SCANF as documented in https://github.com/picolibc/picolibc/blob/main/doc/printf.md:
```
cc -Wl,--defsym=vfprintf=__i_vfprintf -Wl,--defsym=vfscanf=__i_vfscanf
```
If this works, I bet there's a way to get this into picolibc.specs; right now this is done in the `*link` step; maybe LTO uses a different stage?

michalsieron, 2021-08-20:
The thing is LTO seems to change names of the sections and functions:
```
 510:   47d0                    lw      a2,12(a5)
 512:   737f                    0x737f

Disassembly of section .gnu.lto___i_vfprintf.ea67f2ebf25da400:

00000000 <.gnu.lto___i_vfprintf.ea67f2ebf25da400>:
       0:       9c78                    0x9c78
       2:       7c95                    lui     s9,0xfffe5
       4:       7c09                    lui     s8,0xfffe2
       6:       f8f6c74f                fnmadd.s        fa4,fa3,fa5,ft11,rmm
       a:       b99c                    fsd     fa5,48(a1)
```
So I guess it won't work like that. I think I'm left with choice of either defining `PRINTF_LEVEL` like before and having LTO, or doing it picolibc way without LTO.

keith-packard, 2021-08-20:
Yeah, sounds like we need a way to build only the integer path then. That sounds like an easy thing to add.

keith-packard, 2021-08-20:
Give this branch a try and see what you think. https://github.com/picolibc/picolibc/pull/162

keith-packard, 2021-08-20:
We need to build the other printf versions for gcvt, gcvtf, strfromd and strfromf, so this plan doesn't remove the various printf versions, it just switches which version is the 'default', exposed through the vfprintf name.

michalsieron, 2021-08-20:
Yeah, this should work. Thanks for quick fix.

I somehow missed 1eba3461fd1664d38be36f3bf21898f889a7d2e3 so had to update that part, but this one compiles and should also work correctly. Size is correct too. Will actually run code after the weekend.

Depending on target arch, some symbols get optimized out.
For example on riscv, I have to recompile and archive manually _tinystdio/exchange.c_ without LTO, because otherwise I get `undefined reference to '__picolibc_non_atomic_exchange_ungetc'`.
But that was present before and isn't really related to this specific issue.

keith-packard, 2021-08-20:
I bet that's caused by the __weak_reference hack used to pull that code in when the application doesn't provide a replacement. But that's a bit worrying -- you shouldn't be using that function at all unless your target doesn't have the atomic instructions?
I'll close this bug when the patch gets merged; it's waiting for testing now.

closed 2021-08-23 by keith-packard