picolibc/picolibc #223: using picolibc with gnu toolchain

laoshaw, 2021-12-02:
I have gcc-riscv64 built with newlib, can I link my code with picolibc(instead of newlib) using the same gcc/newlib toolchain? 

or do I need build gcc(or clang) with picolib first, then use it to link picolib?

yashi, 2021-12-05:
> can I link my code with picolibc(instead of newlib) using the same gcc/newlib toolchain?

Yes.  I've been using non-picolibc aware GCC to link my app against picolibc and it's been working well.  Picolibc offers spec file you can use for linking.  See https://github.com/picolibc/picolibc/blob/main/doc/linking.md for more detail.

keith-packard, 2021-12-17:
For C applications, this works great. Just build and install picolibc and you'll be able to use --specs=picolibc.specs on the GCC command line to switch to picolibc.

For C++ with libstdc++, you'll need to use a GCC version that has libstdc++ built against picolibc. Support for this is in upstream GCC.

Alternatively, you can use crosstool-ng, which has picolibc support *and* libstdc++ support all included.

closed 2021-12-17 by keith-packard