picolibc/picolibc #232: Restored Docker image is not always the latest

yashi, 2021-12-19:
While searching for a way to auto update docker image when `package.txt` is update (mentioned in #221), I noticed that not all workflow is using the latest docker image generated by the cache maker.

As of writing, [this](https://github.com/picolibc/picolibc/actions/runs/1596785814) workflow, run around Sun, 19 Dec 2021 03:22:28, is the top of the [list](https://github.com/picolibc/picolibc/actions).  In the "Load and Check the Docker Image" you see:
```
Sun, 19 Dec 2021 03:22:28 GMT Run docker load -i $IMAGE_FILE
Sun, 19 Dec 2021 03:23:41 GMT Loaded image: picolibc:latest
Sun, 19 Dec 2021 03:23:41 GMT REPOSITORY   TAG       IMAGE ID       CREATED          SIZE
Sun, 19 Dec 2021 03:23:41 GMT picolibc     latest    2971dfe34d3a   54 minutes ago   5.06GB
```

This image is generated by [this workflow](https://github.com/picolibc/picolibc/runs/4571048170?check_suite_focus=true) run at Sun, 19 Dec 2021 02:34:26 GMT.  So far so good.

However, [this workflow](https://github.com/picolibc/picolibc/runs/4573022324?check_suite_focus=true) doesn't use the latest image even though the workflow run after the the workflow above (02:34:26 GMT vs 06:44:44 GMT in the same day):

```
Sun, 19 Dec 2021 06:44:44 GMT Run docker load -i $IMAGE_FILE
Sun, 19 Dec 2021 06:45:46 GMT Loaded image: picolibc:latest
Sun, 19 Dec 2021 06:45:46 GMT REPOSITORY   TAG       IMAGE ID       CREATED      SIZE
Sun, 19 Dec 2021 06:45:46 GMT picolibc     latest    f99bcfa16bc5   6 days ago   5GB
```

The image was built by [this workflow](https://github.com/picolibc/picolibc/actions/runs/1571568120) at Dec 13th.
```
Mon, 13 Dec 2021 06:15:06 GMT #8 exporting manifest sha256:85b1a685057e7fccae737ba04af2f0c49d7a60f82857b1c4c9363a279a635d40 done
Mon, 13 Dec 2021 06:15:06 GMT #8 exporting config sha256:f99bcfa16bc54f57eaf911241a540400d77315509cb6d0501f6f0857cb28dfd6 done
```

yashi, 2021-12-19:
Did it hit this restriction?
https://docs.github.com/en/actions/advanced-guides/caching-dependencies-to-speed-up-workflows#restrictions-for-accessing-a-cache

keith-packard, 2021-12-19:
Can we make the cache only run from main when packages.txt changes, or maybe once a week otherwise? It's not like we need to track debian packages any closer than that; the only reason for daily updates was to catch changes to packages.txt. And, I'd be happy to get changes to packages.txt merged to main when necessary.

yashi, 2021-12-19:
Let me try.

keith-packard, 2021-12-19:
thanks. it's still using random images even when building main.

yashi, 2021-12-19:
I just noticed that https://github.com/picolibc/picolibc/actions/runs/1599279981

yashi, 2021-12-19:
Still not tested but this is what I came up with
https://github.com/yashi/picolibc/commit/5b641be886e3c9e583a67fe08bda954a39ca5f59
I don't know it works or not yet.

One thing I'm sure you want to do but I haven't figured out is that to make all tests depends on the cache maker when and only when `.github/packages.txt` is updated.  We can, as you can see in the above diff, trigger a workflow with path filter.  But, I can't find path filter which I can apply to jobs, especially `if` expression: https://docs.github.com/en/actions/learn-github-actions/workflow-syntax-for-github-actions#jobsjob_idif

keith-packard, 2021-12-20:
Hrm. Would it help if we used ${{ hashFiles('.github/packages.txt')}} instead of hashing the image file? That way we'd could look for a build that exactly matches the packages.txt contents instead of not really knowing which name to use?

yashi, 2021-12-20:
Thing is that Debian base system changes regardless of the `packages.txt`.  If we don't care about the updates of base packages, Yes, we can use the hash of the file.  Do you want to try?

The reason I'm using a hash value is to generate a unique key. Nothing else.

keith-packard, 2021-12-21:
> Thing is that Debian base system changes regardless of the `packages.txt`. If we don't care about the updates of the base packages, Yes, we can use the hash of the file. Do you want to try?

Right, I don't think we care about short-term changes to the Debian base operating system.We'll pick up changes slowly as the cache entries expire or other changes happen. I've got another attempt running this morning -- trying to embed the cache generation into the individual workflows and skipping the docker image creation steps when the result is already in the cache. That won't be as good as your approach in reducing duplicate image builds, as two workflows running at the same time will still end up creating duplicates, but having explicit dependencies should make things more predictable?

yashi, 2021-12-21:
Thanks and apologize for the troubles.  I couldn't figure out how the cache system works.  Hope it works.  :crossed_fingers:

keith-packard, 2021-12-21:
I think we're in about the same state -- just try things and see what happens. Thanks for exploring the caching stuff; when it does work, it saves a huge amount of time and load on the package sources. I suspect we've both learned a lot here, which is always good.  I really do appreciate the work you've put into improving the picolibc CI environment!

keith-packard, 2021-12-23:
Ok, I think we're back to working, although things aren't as efficient as we had hoped to make them. Every workflow now checks to see if the docker image needs to be rebuilt by checking out picolibc and hashing packages.txt, skipping the build if it isn't needed.

closed 2021-12-23 by keith-packard

yashi, 2021-12-23:
Congrats!
Well, 30 sec for cache-maker prio to 120 tests is reasonably fast IMO.  Now the current time consumer is docker image loading.  It takes 60 sec for each tests.  So, with 120 tests in a workflow, it spends 2hr just loading.

A minsize test is about 15 to 30 sec.  It might be better to run multiple tests on a loaded docker without using Actions matrix.  But, that's another story I suppose.

Thank you again for the cache work!

keith-packard, 2022-02-10:
It *was* another story -- I've gone ahead and created a script that will run multiple targets from the same test matrix element, and that saved about an hour of build time. That gets us to 3 hours for a full run, which seems tolerable. We could extend this to run multiple configurations too; just needs a bit of scripting. That could flatten each of the jobs that get run into something of similar length; the ARM and RISC-V multilib builds each take about 15 minutes, while even the merged builds that this introduces are only taking a couple of minutes each.

yashi, 2022-06-27:
Cache delete is implemented!
https://github.com/actions/cache/issues/2#issuecomment-1167303737