picolibc/picolibc #237: Where should __assert_func() be located?

yashi, 2021-12-21:
This is related to #177.

Where should we put `__assert_func()`? Under Zephyr or picolibc?  Or perhaps app should have it?  There are a few `__assert_func()` under picolibc.  But I feel like Zephyr should have it as a glue code.  Is it OK for picolibc that if I put the function under Zephyr?

keith-packard, 2021-12-23:
I can imagine that zephyr would like to replace __assert_func with something that works better in that environment. Any reason picolibc can't continue to provide a default too?

yashi, 2021-12-23:
Yes, I think Zephyr want to connect to its logging facility.

> Any reason picolibc can't continue to provide a default too?

It can, I suppose.  But, in order to print messages, picolibc must know how to do it.  `printk()`, `printf()`, or `LOG_ERR()`, right?

keith-packard, 2021-12-24:
for bits inside picolibc, that'd be printf, or perhaps fprintf(stderr, ...). If you want to replace this function in zephyr, then you'd probably want to use LOG_ERR?

yashi, 2021-12-24:
I'll have it in Zephyr repo.  Thanks.

closed 2021-12-24 by yashi