picolibc/picolibc #246: Relocations emitted for __data_size/__bss_size symbols for m68k

tdaede, 2022-02-05:
When building a m68k binary with -Wl,q and the generic crt0.c/crt0.h, I get relocation entires emitted for symbols that should be integers and not addresses, e.g:
```
RELOCATION RECORDS FOR [.text]:
OFFSET   TYPE              VALUE 
00000002 R_68K_32          __data_size
00000008 R_68K_32          __data_source
0000000e R_68K_32          __data_start
00000014 R_68K_32          memcpy
0000001a R_68K_32          __bss_size
00000022 R_68K_32          __bss_start
```
This can be worked around by converting crt0.c/.h to use __data_start and __data_end rather than __data_size, however I don't know if there are other solutions that could work without breaking compatibility with existing linker scripts.

(I am using a platform where the loader does the final relocations, hence why I'm emitting them)

keith-packard, 2022-02-05:
I thought it was kinda clever to have the linker do the subtraction instead of making the application do it at runtime. Is there a different type we should be using for these symbols? If so, do you have any idea how to specify that in the linker script?

tdaede, 2022-02-08:
Yeah, it is nice to have a slightly smaller crt0. I've tried swapping the extern type in crt0.c and playing with ABSOLUTE() in the linker script to no avail. I did also find https://stackoverflow.com/questions/42638182/using-linker-symbol-from-c-code-as-a-fixed-constant-not-relocated-in-a-share which ran into a similar problem.

Also on my todo list is to test a clang cross build and see if it's similar. I don't think there is any way for the linker to fix this as the linker only sees indistinguishable relocs from the compiler.

keith-packard, 2022-02-08:
Sounds like we might just need a platform-specific option that does the subtraction at runtime instead of link time. A bit frustrating, but alternatives that involve magic object file hacks at link time don't seem terribly appealing to me.

PR #257 (2022-02-19): Compute memory region sizes at runtime
This series adds a -Dcrt-runtime-size=true option which tells picolibc to compute the size of the bss, data, tbss and tdata segments at runtime by subtracting addresses rather than having the linker perform that subtraction at link time. This can be necessary if the toolchain doesn't deal with relocating the size values as 'addresses'.

Closes #246
closed 2022-02-19 by keith-packard