picolibc/picolibc #270: Meson 0.53.2: Compiler arm-none-eabi-gcc can not compile programs

eugene-so, 2022-04-05:
FYI, I found this issue using meson 0.53.2.  It is fixed in meson 0.54.2, so you might want to update this repo.

---

The Meson build system
Version: 0.53.2
Source dir: /home/eso/build/picolibc
Build dir: /home/eso/build/picolibc/scripts/build-arm
Build type: cross build
Project name: picolibc
Project version: 1.7.6
C compiler for the build machine: cc (gcc 9.4.0 "cc (Ubuntu 9.4.0-1ubuntu1~20.04.1) 9.4.0")
C linker for the build machine: cc ld.bfd 2.34

meson.build:35:0: ERROR: Compiler arm-none-eabi-gcc can not compile programs.

---

Here is a related link that describes the same problem.

https://github.com/mesonbuild/meson/issues/6594

keith-packard, 2022-04-06:
Did you try adding 'skip_sanity_check=true' to the [properties] section of your cross file?

eugene-so, 2022-04-07:
Yes this property was present when the error occurred.

keith-packard, 2022-04-07:
argh. so we can only use the simpler exe-wrapper on newer versions of meson. That's super helpful. I guess we can either revert the exe-wrapper change or bump the required version of meson. I think I can come up with a version that is only a bit more complicated than the current one, but not as complicated as the original one.

keith-packard, 2022-04-07:
If you can give PR #271 a try (branch meson-53-support) to verify whether I'm correct about the failure, that would be super helpful.

eugene-so, 2022-04-07:
The failure still occurs: 

eso@eso-OptiPlex-7080:~/build/picolibc/scripts/build-arm$ ../do-arm-configure -Dc_std=c99
The Meson build system
Version: 0.53.2
Source dir: /home/eso/build/picolibc
Build dir: /home/eso/build/picolibc/scripts/build-arm
Build type: cross build
Project name: picolibc
Project version: 1.7.6
C compiler for the build machine: cc (gcc 9.4.0 "cc (Ubuntu 9.4.0-1ubuntu1~20.04.1) 9.4.0")
C linker for the build machine: cc ld.bfd 2.34

meson.build:35:0: ERROR: Compiler arm-none-eabi-gcc can not compile programs.

A full log can be found at /home/eso/build/picolibc/scripts/build-arm/meson-logs/meson-log.txt
eso@eso-OptiPlex-7080:~/build/picolibc/scripts/build-arm$ git status
On branch meson-53-support

keith-packard, 2022-04-07:
Thanks. I created a python venv to install meson 0.53.2 and reproduced your issue. The fix is to re-add the `-nostdlib` option in the compiler configuration in the cross file as meson 0.53.2 doesn't use any other c flags for early compiler tests. I pushed this change to the meson-53-support branch.

PR #271 (2022-04-09): scripts: Support meson 0.53.2 by adjust exe_wrappers
Meson version 0.53.2 (at least) doesn't check for the
skip_sanity_check value before attempting to run test fragments during
configuration. That means the exe_wrapper value needs to handle being
run at that point. To support this, the exe_wrapper becomes a script
which checks for a environment variable (PICOLIBC_TEST) and exits
indicating success if that is not present.

Closes #270
closed 2022-04-09 by keith-packard