picolibc/picolibc #272: Build fails when -Dnewlib-multithread=false defined

eugene-so, 2022-04-07:
In file included from ../../newlib/libc/tinystdio/stdio_private.h:34:0,
                 from ../../newlib/libc/tinystdio/filestrputalloc.c:36:
../../newlib/libc/tinystdio/stdio_private.h: In function '__posix_lock_init':
../../newlib/libc/tinystdio/stdio_private.h:127:41: error: 'struct __file_posix' has no member named 'lock'
  __lock_init(((struct __file_posix *) f)->lock);
                                         ^
../../newlib/libc/include/sys/lock.h:42:49: note: in definition of macro '__lock_init'
 #define __lock_init(lock) __retarget_lock_init(&lock)
                                                 ^~~~
../../newlib/libc/tinystdio/stdio_private.h: In function '__posix_lock_close':
../../newlib/libc/tinystdio/stdio_private.h:132:42: error: 'struct __file_posix' has no member named 'lock'
  __lock_close(((struct __file_posix *) f)->lock);
                                          ^
../../newlib/libc/include/sys/lock.h:46:50: note: in definition of macro '__lock_close'
 #define __lock_close(lock) __retarget_lock_close(lock)
                                                  ^~~~
../../newlib/libc/tinystdio/stdio_private.h: In function '__posix_lock':
../../newlib/libc/tinystdio/stdio_private.h:137:44: error: 'struct __file_posix' has no member named 'lock'
  __lock_acquire(((struct __file_posix *) f)->lock);
                                            ^
../../newlib/libc/include/sys/lock.h:50:54: note: in definition of macro '__lock_acquire'
 #define __lock_acquire(lock) __retarget_lock_acquire(lock)
                                                      ^~~~
../../newlib/libc/tinystdio/stdio_private.h: In function '__posix_unlock':
../../newlib/libc/tinystdio/stdio_private.h:142:44: error: 'struct __file_posix' has no member named 'lock'
  __lock_release(((struct __file_posix *) f)->lock);
                                            ^
../../newlib/libc/include/sys/lock.h:59:54: note: in definition of macro '__lock_release'
 #define __lock_release(lock) __retarget_lock_release(lock)
                                                      ^~~~
[1086/22802] Compiling C object 'newlib/li...umb_v8_m_main_fpv5_sp_hard@sta/memchr.c.o'
ninja: build stopped: subcommand failed.

keith-packard, 2022-04-07:
We just discovered this issue last week -- rdiez discovered that `newlib-multithread` and `newlib-retargetable-locking` must match or the build fails. This was fixed in #269; are you still having trouble with the 'main' branch?

eugene-so, 2022-04-07:
Yes I am working from the main branch currently.  Should I move to another branch?

On another note, I am just trying to prototype this library on a single thread bare metal system that is already implemented successfully using nanolib.  All our IO goes to and from a uart.  Is there any sample code on how to quickly get this up and running?  THanks!

keith-packard, 2022-04-07:
Here's a complete example for the ARM MPS2-AN385 (the cortex-m3 board supported by qemu).

https://github.com/picolibc/simple-mps

eugene-so, 2022-04-07:
Thank you.  I tested and it looks like this issue is resolved.  I'll check out those links you sent.

keith-packard, 2022-04-07:
Awesome!

closed 2022-04-07 by keith-packard