picolibc/picolibc #273: Thousands separators with tinystdio

rdiez, 2022-04-09:
With tinystdio, `printf "%'d"` yields `%'d` , which is not ideal.

According to the man page for PRINTF(3):

`The POSIX locale uses '.' as radix character, and does not have a grouping character.`

The example shows then how the request to use thousands separators is just ignored. Later on in the same page:

`'  For  decimal  conversion  (i, d, u, f, F, g, G) the output is to be grouped with thousands' grouping characters if the locale information indicates any.`

Therefore I would:
1) Document that neither tinystdio nor legacy stdio support thousands separators.
2) Modify tinystdio so that it ignores the request but still prints the number.

Of course, implementing thousands separators would be nice, even if it is fixed to the English locale, but that is probably a lot of work.

rdiez, 2022-04-09:
I investigated further, and I found out that Newlib's stdio does support thousands separators.

You need to set 'thousands_sep' and 'grouping' first, see localeconv(). Note that setlocale() has no effect if you disable multibyte support with --disable-newlib-mb (which you normally do), but modifying those 2 members directly on the structure returned by localeconv() seems to work (even though it is normally discouraged).

keith-packard, 2022-04-09:
I've never heard of this feature before! Thanks for pointing it out. Not sure how much work it would take to support, but just ignoring it would be pretty easy. This, along with positional args, appears to be the two significant additions to printf that are in POSIX but not ISO C.

keith-packard, 2022-04-10:
oh, looks like the C locale doesn't have a thousands separator, so if that's all we want to support, it will be super cheap. Could add actual code when built with real locale support, but frankly C locales are so terrible it's hard to get excited...

rdiez, 2022-04-10:
Yes, it looks like, strictly speaking, the C or POSIX locale does not have a thousands separator. That is why, by default, Newlib does not print thousands separators when you ask for them.

But that's silly. If you ask for thousands separators, it does not make sense to ignore the request. After all, you want separators because long numbers are hard to read.

I am guessing the reason why you can call localeconv() and modify those values directly is to get such features while saving a lot of locale work.

This is the test code I kept for future reference:

```
// Thousands separators, as in printf( "%'d" ), are supported by Newlib's stdio, but you need to set
// 'thousands_sep' and 'grouping' first, see localeconv(). Note that setlocale() has no effect if
// you have disabled multibyte support with --disable-newlib-mb (which you normally do), but modifying
// those 2 members directly on the structure returned by localeconv() seems to work
// (even though it is normally discouraged).
#ifdef _MB_CAPABLE
  #error "Newlib's multibyte support is enabled (see --enable-newlib-mb). I need to investigate if setting 'thousands_sep' and 'grouping' can be done in the normal locale way."
#endif

struct lconv * const localeInfo = localeconv();

// You are normally discouraged from changing the lconv members, but we make an exception for Newlib.
// The strings we are passing are 'const', but that should be OK. I do not expect anybody to change them in place.
localeInfo->thousands_sep = const_cast< char * >( "," );
localeInfo->grouping      = const_cast< char * >( "\x03" );

// Afterwards, this yiels "123,456,789" as expected:
//   sprintf( buffer, "%'d", 123456789 );

```

I am guessing that such minimal support for thousands separators would not be hard to implement for tinystdio.

Alternatively, I think that a fixed (compilation time) character for thousands separators, and a fixed grouping of 3, would satisfy 99% of the users.

keith-packard, 2022-04-10:
> But that's silly. If you ask for thousands separators, it does not make sense to ignore the request. After all, you want separators because long numbers are hard to read.

This all seems very odd.

 * ISO C includes the thousands_sep member of the struct, but does not use it in any API that I can find.
 * POSIX added the ' flag to printf, but then doesn't include a thousands separator in the POSIX locale; presumably because the POSIX locale is defined to be the C locale, which specifies an empty string.
 * ISO C makes it clear that modifying the results from localeconv() results in undefined behavior.
 * 
> I am guessing the reason why you can call localeconv() and modify those values directly is to get such features while saving a lot of locale work.

Not in any standard I can find -- ISO C says

    "The localeconv function returns a pointer to the filled-in object. The structure pointed to by the
     return value shall not be modified by the program"

> I am guessing that such minimal support for thousands separators would not be hard to implement for tinystdio.
> 
> Alternatively, I think that a fixed (compilation time) character for thousands separators, and a fixed grouping of 3, would satisfy 99% of the users.

I'm very hesitant to add features to picolibc which require applications violate the standards to which picolibc is expected to conform, so anything which requires writing to the return from localeconv seems like a bad idea.

But, I'm also sensitive to requests to provide reasonable functionality, and having the ' flag to printf do nothing also seems silly.

As the C locale is defined with an empty thousands_sep value, we can't really make a picolibc redefine that locale as that would violate a required element of the standard. We can complain about the standards all we like, but violating them for personal preference isn't something I'm interested in.

Locale support in picolibc is not great overall; most of the support was (as with much of newlib) written with only cygwin in mind. Updating that to support embedded environments that want to provide full locale support will be a big job, and something which I'm not sensing a huge amount of need. Plus, it's a pretty significant chunk of code, so requiring all of that just to get thousands separators in numbers seems a bit much.

In short, I'm not seeing a simple solution here -- the only standards-conforming solution involves adding locale support for advanced locales like en_US.UTF_8. With that, we *could* only define locales which set the thousands_sep to ',' and the grouping to 3, which would at least reduce the amount of code for this feature, but locales include a huge range of changes to the C library.

yashi, 2022-04-11:
For these like me, here is a reference to a c++ discussion on this before adopted to C++14.
http://www.open-std.org/jtc1/sc22/wg21/docs/papers/2013/n3499.html

rdiez, 2022-04-11:
> but violating them for personal preference isn't something I'm interested in

I wouldn't say that providing a compilation-time setting for a separator character that the Picolibc user can configure is "violating" a standard "for a personal preference".

I wouldn't implement full locale support either for a library designed for embedded software. But we could add a new Picolibc-specific API, like PicolibcSetThousandsSeparators( ',', 3 ).

rdiez, 2022-04-11:
> For these like me, here is a reference to a c++ discussion on this before adopted to C++14.

I think that document is about thousands separators in C++ source code. That would be:

printf( "%'d", 123'456'789 );

The compiler should be able to do that already. The issue I raised here is about thousands separators at run-time. I mean that printf( "%'d" ) does not print thousands separators, and in the case of Picolibc, it prints "%'d", instead of "123456789".

keith-packard, 2022-04-12:
Ok, I've created a PR that "fixes" the bug by simply ignoring the thousands separator. To make this actually supported, we'll need to either add more locale support to tinystdio or create a new interface somewhere to control this.

All of this makes me wonder whether it would be interesting to explore supporting a broader set of UTF-8/ASCII locales in the code. Limiting support to a single charset would make it simpler, although we'd still end up needing huge tables to deal with case folding and collation ordering. Hrm.

rdiez, 2022-04-12:
I think that your "fix" is fine. Because you cannot really trust libc developers, I already have my own routine that adds thousands separators anyway. ;-)

closed 2022-04-12 by rdiez

keith-packard, 2022-04-12:
> I think that your "fix" is fine. Because you cannot really trust libc developers, I already have my own routine that adds thousands separators anyway. ;-)

Well, I think you can trust the libc developers to follow the spec, it's just that this specification is quite silly. A printf flag which, by default, has no effect, doesn't seem terribly useful. 

I'd be interested in knowing if you think C library locale support that was limited to LC_NUMERIC would be useful? I'd suggest including LC_MONETARY, but we don't have anything which uses the LC_MONETARY fields that I can find (does anything other than strfmon use them?). I'm just not sure including LC_CTYPE and LC_COLLATE just to get LC_NUMERIC is reasonable.

rdiez, 2022-04-12:
I have never need any locale stuff for my embedded software, and I never investigated how to configure it. I am programming rather small microcontrollers though.

The only thing I found useful is the thousands separators in "%'d", and I've never needed anything other than the standard English separators like "1,000,000". I have always found long numbers like "13389741" unreadable, so I ended up implementing my own routine to add thousands separators to integer numbers, which was a little annoying. Had I realised earlier that Newlib can actually do it, it would have saved me quite a lot of work, because I could have shared code that uses sprintf and the like more easily between PC and embedded.

I bet most embedded people would be happy if Picolibc provided some way to set the English ',' to be the thousands separator, so that "%'d" works with a minimum of fuss. Exactly how does not really matter. Real programmers build their own libc, you know. 8-)

Sometimes, you have to cater for the Germans in the user interface, and they have different separators, like "1.000,5". And the Spanish have others too. But most people accept the English separators, especially from embedded software that works on a small microcontroller.

I wasn't even aware that strfmon existed. I never needed to format currency, let alone with different rules than the other numbers.

For me, the ability to set the thousands separators and the decimal point at runtime would be real luxury worth spending some bytes of RAM. This way, even the Germans would be happy. 8-) Most cultures use a fixed grouping of 3.

The trouble is, things like setlocale() seem rather rigid: you cannot specify your separator characters, you have to specify some locale name instead. It is all geared towards config files that the user can edit and even compile/translate. That means having a filesystem, a no-go for small microcontrollers. Following standards is nice and all, but sometimes they just don't cut it.

That's why I would add a new API. We could just add simple global variables like errno, which could be placed in TLS if you have threads (like errno), so that one thread can generate an English user interface while another one can generate a German one. Or one thread generates a (somewhat) localised web page, while another one exports data for machine consumption with English separators.

Something like "libc_thousands_sep" and "libc_decimal_sep" (or "tinystdio_xxx" instead of "libc_xxx") would suffice for the vast majority of users. The rest can install embedded Linux. 8-)

Or perhaps we could take a pointer to a global structure, to be set per thread. That would be more future-proof. The first version of the structure could have just those 2 members. If you need more in the future, the structure could grow. But then you would need a way to clear it, like picolibc_init_locale_structure( &my_structure ), so that older software still works with newer Picolibc versions where the structure has more members.

Another way would be to let the user provide a callback like this: get_separator( THOUSANDS ). The trouble is, the data type to return is sometimes different. For example, the grouping is actually a string, to support more flexible grouping (start with 3, then use 2).

keith-packard, 2022-04-13:
Yeah, that's why I was wondering if having just LC_NUMERIC support could make some sense -- as you know, that defines a struct which can be modified through setlocale(LC_NUMERIC, "locale"). It's not the best API, but it is a standard, and it would provide exactly this functionality -- call `setlocale(LC_NUMERIC, "")` and you'd get the single built-in non-POSIX locale, which could be en_US (with ',' for thousands and '.' for decimal), or 'de_DE' (with '.' for thousands and ',' for decimal). We'd have a few built-in structs with the LC_NUMERIC values. There are defined APIs for making the locale per-thread, and we could even implement those using TLS. Standards-conforming code that builds with picolibc or glibc seems like a pretty reasonable goal, even if the resulting API isn't awesome.

rdiez, 2022-04-13:
I'm not quit following you here, but I don't know much about locale support though, so please bear with me.

LC_NUMERIC does not "define a struct which can be modified through setlocale", as far as I can tell.

setlocale(LC_NUMERIC, "") wouldn't really give you "the single built-in non-POSIX locale", but it is supposed to look in the environment, and I am trying to get rid of environment variable support to save memory. Such environment variables wouldn't provide separator characters etc. anyway, they would be variables with names like LANG or LC_NUMERIC which provide a locale name.  This indirection is not really necessary in an embedded software.

Well, I guess you could make a special case for setlocale(LC_NUMERIC, ""), so that it does switch to the "the single built-in non-POSIX locale". The user has to define the locale when compiling Picolibc though. That's fine with me, I would just make it "en_US" like you suggested, so that thousands separators work with a minimum of fuss. I could then share a lot of printf("%'d") code between PC and firmware. But you wouldn't be able to distribute a binary Picolibc for all users, say like a Debian package, where the user could freely choose the thousands separator.

The next step up would be to have at least 3 locales: The first one would be "C" alias "POSIX", because it is required by the standard. That means no decimal separators. The second one would be "en_US", in case the application needs to generate data that is going to be read by some other software. Such data exchange formats tend to use the English thousands separators and decimal point. The debug console would always use the English settings. The 3rd locale would be the one my users want, like "de_DE". I could add further locales for all the languages my application supports (at Picolibc compilation time).

setlocale( xxx, "" ) must still work, but exactly which locale it picks, I don't really mind, for I would always specify the locale I want.

But this "upgraded" solution would still be unsatisfactory. I would certainly help me now, but you still wouldn't be able to supply a binary for all possible users. The main issue is that the application wouldn't have the freedom to set an arbitrary separator character. I could imagine giving the end-user a configuration setting for the thousands separator, and another configuration setting for the decimal point. With 2 such settings, the firmware can achieve 99 % of the requirements with very little cost. But the setlocale() API is just not able to take any arbitrary separator character. As far as I can tell, the only way is the discouraged method of calling localeconv() and modifying the returned structure.

Or did I miss the part of the standard where you can freely set such characters, that is not discouraged, and which does not require separate storage (like a filesystem) holding the separator characters and other locale settings?

keith-packard, 2022-04-14:
I think you've understood the situation quite well. As to environment variables, I agree that we don't want to add support for those; it's hard to imagine where a tiny embedded system would even store them. We haven't found any way for the user to specify arbitrary thousands and decimal point characters through the standard APIs, and I've read them in some depth.

However, to make those useful, we'd be talking utf-8 values, and that starts to make the code a lot more complicated. There aren't any locales that use ASCII code points other than ( , . ) and ( . , ) in all of the the locales offered by glibc -- the other options are U+066C (arabic ٬), U+202F (narrow non-breaking space), or U+2019 (apostrophe ’). For decimal point, the only other option is U+066B (arabic ٫). So, just offering three options, ( <none> . ), ( , . ), ( . , ) would support all locales using ASCII for these settings.

rdiez, 2022-04-14:
> So, just offering three options, ( none . ), ( , . ), ( . , ) would support all locales using ASCII for these settings.

So that would be 3 locales: C/POSIX, one English locale and say the German one. That is certainly much better than simply ignoring requests to print thousands separators, and it would cover most cases for small embedded devices.

keith-packard, 2022-04-14:
I'll see if I can explore this option then; might be a while, this isn't exactly a requirement for anything I'm doing with picolibc :-)